thonny/thonny #3203: Pico won't connect Device Busy

Leofric17, 2024-05-19:
I have several Picos and all work on Thonny on my desktop version with Win 10.
On my Pi 4 I get error "Device is busy..... "  But, it isn't - nothing running.  No Main.py 

Same cable used for both.

If I open in bootloader mode the Pi 4 recognises the Pico OK.

Try Open File in Thonny and it offers Computer or Pico - selecting Pico gives the error.

I have done a full update, uninstalled Thonny and re-installed but no change.

aivarannamaa, 2024-05-22:
Have you been able to use any MicroPython device on this Pi 4?

Can you get MicroPython REPL when connecting with `tio` (`sudo apt install tio` and `tio /dev/tty<whatever>`)? You may need to press Ctrl+B after connecting.

Leofric17, 2024-05-22:
> Have you been able to use any MicroPython device on this Pi 4?
> 
> Can you get MicroPython REPL when connecting with `tio` (`sudo apt install tio` and `tio /dev/tty<whatever>`)? You may need to press Ctrl+B after connecting.

Thanks for your help.
I have previously been able to use the same Pi4 with Pico and Pico W.

Using 'tio' it says connected.   Not sure how to get beyond that. 
Also if Thonny is running first it says can't connect as you would expect.
If I quit tio and open Thonny I get the "Device busy....." message.  If tio is running Thonny reports that port is being used and it can't connect.

Seems Thonny knows the Pico is there but won't access it.   But the Win 10 machine has no issues.

aivarannamaa, 2024-07-12:
Sorry for the late reply!

> I have previously been able to use the same Pi4 with Pico and Pico W.

In case it's still relevant -- was it the same Pico and how did you use it?