thonny/thonny #3204: Won't save to Pico WH

Magnusretrotech, 2024-05-19:
When I click 'MicroPython (Raspberry Pi Pico) Board in FS mode @ /dev/ttyACM0' at the bottom of the window, It pops up with this in the terminal `
Unable to connect to /dev/ttyACM0: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

Try adding yourself to the 'nfsnobody' group:
> sudo usermod -a -G nfsnobody <username>
(NB! You may need to reboot your system after this!)

Process ended with exit code 1.
`

I'm using Linux Mint 21.3

aivarannamaa, 2024-05-22:
Have you done this:

```
Try adding yourself to the 'nfsnobody' group:
> sudo usermod -a -G nfsnobody <username>
(NB! You may need to reboot your system after this!)
```

?

Magnusretrotech, 2024-05-25:
usermod: group 'nfsnobody' does not exist

aivarannamaa, 2024-07-12:
Sorry about the late reply!

What happens, when you use `dialout` instead of `nfsnobody` as group name?