thonny/thonny #3205: problem in save and open file

Hossamo-5, 2024-05-21:
I use Ubuntu 24 When I want to open a file, the program does not respond, and when I save a code that I wrote, it does not work either. Please solve this problem.

08:06:17.688 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
--help

aivarannamaa, 2024-05-21:
I suppose you have older Thonny version, which does not work with newer Zenity version. 

The easiest option is to turn off Zenity dialogs from "Tools => Options => General".

Hossamo-5, 2024-05-27:
Thanks for solving my problem.

> I suppose you have an older Thonny version, which does not work with the newer Zenity version.
> 
> The easiest option is to turn off Zenity dialogs from "Tools => Options => General".
 
Thanks for solving my problem.

closed 2024-05-29 by aivarannamaa