thonny/thonny #3207: Change directory problems with micropython

asherevan, 2024-05-22:
After changing directories with os.chdir and then trying to change back to the previous directory with the open file dialog in thonny it only shows the contents of the same directory as folders.

aivarannamaa, 2024-07-12:
Thank you for the report and sorry for the late reply!

> it only shows the contents of the same directory as folders.

I didn't quite understand this. Can you rephrase it?

asherevan, 2024-07-12:
Instead of changing directories and displaying the files it just stays in the same directory and shows the files as folders (with no contents) instead.

aivarannamaa, 2024-07-12:
I was not able to reproduce this. Could you please post a screenshot?

asherevan, 2024-07-12:
I’m on a raspberry pi so I couldn’t take a screenshot, but here’s the best picture I could take.

This is after changing directories with os.chdir then trying to change to the home directory in the file browser. And it is only showing the contents of the other directory.

[image]