thonny/thonny #3208: How to turn off find&replace

7CusTanCirus, 2024-05-26:


wtakuo, 2024-05-28:
HI. 
I don't know the exact purpose of this issue, but from the title, I would guess that those who like Emacs-style cursor manipulation probably have a hard time if ctrl-F always invokes Find & Replace and the behavior is not customizable.

I use Thonny on my Mac, which I think is a very good editor, but I was having trouble with this problem. I tried to solve this problem as follows.

This is a rather rough method, but I directly edited the following file in the installed Thonny.app.
`/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/tktextext.py`

What I did was to comment out the following line in the method `_bind_editing_aids` of the class `EnhancedText` in this file.
    ```self.bind("<Control-f>", self._redirect_ctrlf, True)```
This allows ctrl-F to move the cursor forward.  Find & Replace can be invoked with cmd-F without any problem.

The above is clearly ad hoc and symptomatic, and will be a problem on non-Mac platforms. I have been hesitant to post it as an issue until now, but I decided to take this opportunity to write about it here. I wish there was a smarter way.

aivarannamaa, 2024-05-29:
@7CusTanCirus , if you are on macOS and missing the close button in the title-bar, then try pressing Esc on the keyboard.

closed 2024-12-04 by tanbaycu