thonny/thonny #3209: Process ended with exit code -11

deliciousCodeSandwich, 2024-05-29:
I was just working on a project to store birthdays with guizero, and I was checking the guizero website on how to use it (because I am not very experienced with guizero), and a Python message came up that said 'Python quit unexpectedly.' I ignored it, but when I went back to Thonny, it said 'Process ended with exit code -11.' And then I found myself unable to use the run, debug, or any other one of the buttons. I quit Thonny and reopened it (I am on Mac), and the buttons were usable again. It is a minor issue, but annoying.

deliciousCodeSandwich, 2024-05-30:
Update: the same thing happened, except it said 'Process ended with exit code -5.' I also figured out that if you click the stop button, it fixes the run and debug buttons.

aivarannamaa, 2024-07-12:
That's expected -- when a program closes with an error code, Thonny does not continue until you explicitly tell it to stop/restart.

closed 2024-07-12 by aivarannamaa