thonny/thonny #3210: Stop publishing Linux binaries

aivarannamaa, 2024-05-30:
As Linux distributions have become more diverse (more architectures, more differences in core libraries), it is more difficult to produce binaries. I'll rather focus on improving the installation script, which installs necessary system packages, creates a special venv and installs Thonny there.

aivarannamaa, 2024-05-30:
TODO: test, also pay attention to produced uninstaller

sameersharma2006, 2024-06-01:
Although this is done and i understand the reason behind this , i oppose this as apt/deb packaging doesn't have latest thonny versions, snaps with the help of the above binaries was serving as a supplement to provide latest version to Ubuntu & other distros supported by snapd(Debian included).
Hence IMHO the effort/time needed to build the binaries were legitimate & justified.

Even after months of attempts , i have failed to make snap independent of upstream binaries.
`Thonny-5.x +` as a snap won't be available i guess until i sort out this(Seems impossible) & hence i believe once thonny 5.x is published & i am unable to get it snapped, i will retire the snap builds & post a depreciation notice in the forum. 🤷

brianlinuxing, 2024-06-17:
I can appreciate the issue here @aivarannamaa , as a long term Linux user of Thonny, I would say as long as the thonny-x.x.x-86_64.tar.gz is shipped then that’s workable for me. :)

Equally, other solutions** might be to bring on more people, delegate the building of machine specific releases (a common practice on many other projects) or seek automated CI/CD solutions for producing builds where appropriate (large vendors will often offer free credits for open source projectS to do such things, so potentially almost zero cost).

[** You might be doing part or all of these already, it is unclear to me.)