thonny/thonny #3213: (ModuleNotFoundError: No module named 'pkg_resources')

darrellajenkins, 2024-06-08:
Using: Thonny 4.1.3; Windows 11; Python 3.12.3
When attempting to access Tools menu - Manage Packages I get the following error:

PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_active_distributions' (ModuleNotFoundError: No module named 'pkg_resources').
See Thonny's backend.log for more info.
Process ended with exit code 1.

Thonny does not locate any pip install pkg on my computer.  Also I cannot access through Tools - Manage Packages. The box for Search on PyPi in inactive and greyed out. This happens on 2 different windows computers.

aivarannamaa, 2024-06-09:
Thonny 4.1 does not fully support Python 3.12.

I hope to release the first beta of Thonny 5.0 in couple of weeks.

closed 2024-06-09 by aivarannamaa

darrellajenkins, 2024-06-09:
Thank you for your reply.  Thonny is an excellent IDE!