thonny/thonny #3214: Errors message when using a python 3.12 enviroment in my project

kudomarkos, 2024-06-08:
Errors message when using a python 3.12 enviroment in my project:

/usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py:532: SyntaxWarning: invalid escape sequence '\L'
  os.path.expanduser("~\\AppData\Local\Programs\Python\Python3%d" % minor),
/usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py:533: SyntaxWarning: invalid escape sequence '\L'
  os.path.expanduser("~\\AppData\Local\Programs\Python\Python3%d-32" % minor),

aivarannamaa, 2024-09-02:
Sorry for the late reply. Thonny 4 is not guaranteed to work properly with Python 3.12

closed 2024-09-02 by aivarannamaa

ryle2020, 2025-02-06:
please edite the "/usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py" and modify the line 532 and 533:
from 
os.path.expanduser("~\\AppData\Local\Programs\Python\Python3%d" % minor),   
os.path.expanduser("~\\AppData\Local\Programs\Python\Python3%d-32" % minor),
to
os.path.expanduser(**r**"~\\AppData\Local\Programs\Python\Python3%d" % minor),   
os.path.expanduser(**r**"~\\AppData\Local\Programs\Python\Python3%d-32" % minor),

kudomarkos, 2025-02-08:
sorry but the cpython folder doesn't exist in thonny 4.1.7 in a python  3.12 enviroment.
i had open the three cpython-something/__init__.py files.. and expanduser is not there..

[image]

ryle2020, 2025-02-12:
Hi, from your program's Error Message: " /usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py:**532**: SyntaxWarning: invalid escape sequence '\L'
  os.path.expanduser("~\\AppData\Local\Programs\Python\Python3%d" % minor),"
 You must have the file /usr/lib/python3/dist-packages/thonny/plugins/cpython/__init__.py with 532+ lines in there. 

What is your Os, Windows or Linux? Could you go to /usr/lib/python3/dist-packages/thonny/plugins to see if any director named cpython