thonny/thonny #3215: Added Ctrl + / to custom shortcuts for comment toggling, but the shortcut also selecting current line

mgbdevp, 2024-06-09:
The problem is in the title, thonny's default Ctrl+3 comment toggling is working well but i can't get used to this. Wanted to change shortcut to Ctrl+/ and this happen. Also, Ctrl+/ always selecting the line even using Ctrl+3 for comment toggling.

silverleaf21, 2024-07-31:
I have a similiar problem (german qwertz keyboard layout):

- If I have no custom shortcut, all lines of code are selected with ctrl+/ (ctrl-shift-7)
- If I have Control-Shift-Key-7 as shortcut, same happens but  no comment toggling
- If I have Control-slash as shortcut, all lines are selected and than commented/uncommented (so both selecting and toggling)

I chose to use Control-Key-7 for now - this is close enough and works as expected.

aivarannamaa, 2024-09-03:
Please see https://github.com/thonny/thonny/wiki/Custom-shortcuts

silverleaf21, 2024-09-03:
Thanks for your reply. It would be nice it you could point out how this wiki entry should help?

I am able to set a custom shortcut for "toggle_comment". I tried different keys and Control-slash and Control-Key-7 both work. But the former also does select all lines of code - regardless if custom shortcuts are present or not.

Do you want us to tell that we should also rebind the "select all" command? 
Because I can tell you it does not seem to work. 

Ctrl+/ always seems to select all lines of code.

silverleaf21, 2024-09-03:
Short update:
While I was writing I got the info that you also answered on #3270 

> I'll investigate this.

As that is the exact same topic as described here (at least to my understanding): Can I assume that this answer also applies to this issue and it is a bug of some sort?

silverleaf21, 2024-09-04:
I've never worked with Tk before - that's interesting to know. Thanks for clarifying.