thonny/thonny #3217: Crash Report from program

SpudGunMan, 2024-06-13:
version 3.3.1.4, program reported the following, OS is latest mint on intel chip

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/usr/lib/python3.10/tkinter/__init__.py", line 839, in callit
    func(*args)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2139, in poll
    on_completion(proc, out_lines, err_lines)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pylint/__init__.py", line 92, in _parse_and_output_warnings
    check = checks_by_id[atts["msg_id"]]
KeyError: 'R0022'

```

aivarannamaa, 2024-09-02:
> version 3.3.1.4

Can you re-check the version? There is no such Thonny version

SpudGunMan, 2024-09-03:
How about 3.3.14

aivarannamaa, 2024-09-03:
3.3.14 does exist. If you have this version, then I recommend upgrading to 4.1.4