thonny/thonny #3226: thonny python with micro python

franzstr, 2024-06-20:
I have installed micro python on my laptop and connect raspberry pi pico to the laptop. 
the connection is okay, but as I run my script, starting with "import machine" no module named " machine" appears.
The same script works correctly, when I connect  pico to my raspberry pi. 
I am 76 years old and not a profi, but I want to work a bit with python together with my grandson, who is nearly 4.
For Your help I thank you very much.
Franzstr

aivarannamaa, 2024-09-02:
Sorry for the late reply. Are you still having this problem?

franzstr, 2024-09-12:
Hello Aivar,
thank you for being present to help.
My mistake was, not to work with the correct interpreter. I have changed this via raspiios -bullseye and it works.because here the interpreter is fix installed.
I know, I have very much to learn with python, and I would be happy to stay in contact with you as a total beginner.Thank you
Franz,
and sorry or my late reply, I hat a crash with my laptop