thonny/thonny #3241: Thonny Open, Save, Save To ,... don't work

RSMs007, 2024-07-05:
Hi I'm use Ubuntu 24.04 LTS with Thonny 4.0.1

When creating a program in Thonny, I cannot save it!!
Same for Save To, Save, Open, .... buttons
At least those buttons are lit.

Thanks

aivarannamaa, 2024-09-02:
Sorry for the late reply!

It can be that the file dialog has gone behind Thonny window. Try force-closing Thonny and then switching off Zenity dialogs in Tools => Options => General

digizun, 2024-09-03:
Yes turning off Zenity worked for me.
Otherwise the error reads: 
WARNING thonny.ui_utils: Zenity returned code 255 and stderr  ...
Ubunty 24.04 currently uses zenity/noble,now 4.0.1-1build3 and thonny/noble,noble,now 4.0.1-1 
Maybe one of these needs to be upgraded

closed 2024-10-12 by aivarannamaa