thonny/thonny #3243: Alternate Hotkey to run the code

GitMalk, 2024-07-07:
As CTRL+Enter or Shift+Enter are the common hotkeys to execute python code: please provide a setting option for changing the run-command from F5 to an other hotkey combination.

Hermann12, 2024-07-07:
You can setup by your own, see https://github.com/thonny/thonny/wiki/Custom-shortcuts!

GitMalk, 2024-07-07:
Thank you Hermann!

> You can setup by your own, see https://github.com/thonny/thonny/wiki/Custom-shortcuts! ( i.g. **E** for execute)
> 
> > [shortcuts]
> > run_current_script = <Control-E>
> 
> Add this two lines on the end of your **configuration.ini** and you can use **Ctrl+Shift+E** to run your current opened script instead of **F5**.

closed 2024-07-07 by GitMalk