thonny/thonny #3247: Can't upgrade or uninstall packages on Thonny 4.1.4

Spaghettatron, 2024-07-09:
Using Thonny 4.1.4 64bit on Windows 10, I can't upgrade or uninstall any plugins or packages.

[image]

Hermann12, 2024-07-11:
> Using Thonny 4.1.4 64bit on Windows 10, I can't upgrade or uninstall any plugins or packages.
> 
> 
> 
> [image]

You can go to __Tools__ | __Open system shell..__ than

`$ pip install --upgrade <package-name>
$ pip install -U <package-name>`

or 

`$ pip uninstall <package-name>`

aivarannamaa, 2024-09-02:
Package management will be greatly refactored in Thonny 5. Please reopen if the problem remains.

closed 2024-09-02 by aivarannamaa