thonny/thonny #3250: Text input is very slow

wohltat, 2024-07-12:
For some unknown reason the text input got super sluggish recently in Thonny. It takes a while after the keypress is recognized.

[image]

It can also happen that while scrolling with arrow keys the text cursor is so slow that i have to wait several seconds before i can start typing again.
It worked well before and i don't really know what to do about it.

I tried it on a different ubuntu machine and it works fine. But in other editors on my Manajaro machine i don't see such a behavior so it's something about Thonny.

---------------
Thonny 5.0.0b1-dev (latest git version)
Manjaro Linux
Python 3.12.4
Tk 8.6.14

aivarannamaa, 2024-07-12:
Can you try again an older version, which worked well? It would be a great help if you could find out the specific commit, which introduced this problem (perhaps with `git bisect`).

Without this I don't have any ideas where to look, because I haven't touched Thonny's editor for a while.

wohltat, 2024-07-12:
I went back as far as 3.3.0b7-dev and it was still present. So it seems to be only indirectly connected to Thonny.

It seems to something about a blocking IO operations since i notice hard drive activity when typing or moving the mouse cursor.

aivarannamaa, 2024-07-12:
Thanks for the investigation?

Some people have experienced decline in performance when upgrading their Raspberry Pi-s to Bookworm (#3036). Perhapse there is something similar going on with Manjaro.

Can you please check, whether Python's IDLE (which uses the same GUI toolkit, as Thonny) has same problems?

wohltat, 2024-07-12:
Indeed IDLE and also MrPython (which is based on IDLE) show this sluggish key input behavior.