thonny/thonny #3253: ModuleNotFoundError: No module named 'tkinter' (thonny_v)

cmesoar, 2024-07-13:
1) Initiated virtual environment under Mint 21.3 {system info here:} [https://termbin.com/rn6f](https://termbin.com/rn6f)
2) installed Thonny [pip3 install thonny] successfully {see terminal output here:} [https://termbin.com/poli
](https://termbin.com/poli)
3) launch Thonny and get the error: ModuleNotFoundError: No module named 'tkinter' (thonny_v)
      {terminal output from Thonny:} [https://termbin.com/2yv5](https://termbin.com/2yv5)

Quite likely something I am doing/not doing - likely something to do with setting a path variable or config file.
Thanks for any assist!

vsquared, 2024-07-13:
Do you have python installed?  tkinter is included with python.

cmesoar, 2024-07-13:
Standard install of python3 in Linux Mint...so the venv should be standard - couldn't operate a venv wtihout python already installed as far as I know...

cmesoar, 2024-07-14:
more to the venv...according to realpython, the venv is supposed to provide access to the installed python environment: [https://realpython.com/python-virtual-environments-a-primer/#it-links-back-to-your-standard-library](https://realpython.com/python-virtual-environments-a-primer/#it-links-back-to-your-standard-library)

> The Python executable in your virtual environment has access to the standard library modules of the Python installation on which you based the environment. Python makes this possible by pointing to the file path of the base Python executable in the home setting in pyvenv.cfg:

> 

Any other suggestions / ideas? Thanks!

cmesoar, 2024-07-14:
ok....thank you again...your comment did direct me to the solution....after much searching and reading many posts along with helpful assistance from LEO AI...

tkinter is no longer installed by default in linux mint 21.3 cinnamon

to install it in a venv requires that you know it is python3-tk module...
                         again much searching to find that....

finally must know that it is NOT a PIP install but `sudo apt-get install python3-tk` 

and if you wish to have it only in the virtual then that command should be run AFTER the venv is activated...

now Thonny launches perfectly in the venv

hopefully this will help other newbs like me :)

cmesoar, 2024-07-14:
solve

closed 2024-07-14 by cmesoar