thonny/thonny #3255: ReverseShell Virus notice on Thonny.

Tbilgere, 2024-07-21:
Recently updated to the most recent version of Thonny and was greeted with a virus notice during my weekly scan. 

Can someone cross check and or confirm this is a false positive? 

[image]

aivarannamaa, 2024-07-21:
Thank you for the report!

Executable `undill` comes from Python package `dill`, which is a requirement for `pylint`, which Thonny uses for finding possible problems in code.

This is how the executable looks like 

```python
#!/bin/sh
"exec" "`dirname $0`/python3.10" "$0" "$@"

#
# Author: Mike McKerns (mmckerns @caltech and @uqfoundation)
# Copyright (c) 2008-2016 California Institute of Technology.
# Copyright (c) 2016-2023 The Uncertainty Quantification Foundation.
# License: 3-clause BSD.  The full license text is available at:
#  - https://github.com/uqfoundation/dill/blob/master/LICENSE
"""
unpickle the contents of a pickled object file

Examples::

    $ undill hello.pkl
    ['hello', 'world']
"""

if __name__ == '__main__':
    import sys
    import dill
    for file in sys.argv[1:]:
        print (dill.load(open(file,'rb')))
```

Unfortunately the library method which gets called (`dill.load`), is not trivial to review, but I suppose it got flagged, because the library uses Python built-in procedure called `exec`, which takes a string of Python code and runs it. This makes `exec` a useful tool for malware creators. Also, if not used carefully, the power of this procedure can increase the risk of being exploited. I checked https://www.exploit-db.com/, but it does not say anything about "dill".

I assume this is a false alarm, because Pylint is a hugely popular package and if there was something fishy with it or with its dependencies, it would have made to healines by now (the version of dill, which is bundled with Thonny 4.1.4 was released [one year ago](https://pypi.org/project/dill/0.3.7/)).

Tbilgere, 2024-07-21:
@aivarannamaa Thank you so much for your very detailed response! Seeing as the file in question was located in the Python bin I assumed this was either a false positive or maybe possibly a file introduced through a supply chain exploit. Since you have tracked the this files introduction to `pylint` and there is no obvious issue there, I think your assumption that this is a false alarm is a good one. Thank you again for your attention and insight.

closed 2024-07-21 by Tbilgere

aivarannamaa, 2024-07-23:
Nevermind, this page was for businesses. I now reported the false alarm under https://www.bitdefender.com/consumer/support/answer/29358/

aivarannamaa, 2024-07-27:
@Tbilgere, can you please check, whether BitDefender still reports https://github.com/thonny/thonny/releases/download/v4.1.4/thonny-4.1.4.pkg as malware?

3 days ago I got response from BitDefender:

> Our Malware Research Team has finished analyzing your case.
> The file is clean and currently not detected by our engines.

but https://www.virustotal.com/gui/file/3acf16c0111fbecf351bd0c9d15e7cc0db06578699dac930ac3f9de19c6b0c8d/detection still shows that BitDefender has flagged it.

At the moment I don't know if VirusTotal hasn't updated their definitions or I need to get back to BitDefender support.

aivarannamaa, 2024-09-03:
Thonny is now green again in VirusTotal (https://www.virustotal.com/gui/file/3acf16c0111fbecf351bd0c9d15e7cc0db06578699dac930ac3f9de19c6b0c8d/detection)