thonny/thonny #3258: VirusTotal Alerts for Thonny 4.1.4 macOS

paul-cossey, 2024-07-23:
Hi, Folks

https://www.virustotal.com has just started to flag Thonny as potential Malware. I'm sure these are false positives, but the most recent scan just now has five vendors flagging it. 

[image]

This is a direct link to page: https://www.virustotal.com/gui/file/3acf16c0111fbecf351bd0c9d15e7cc0db06578699dac930ac3f9de19c6b0c8d/detection (or if you don't trust random internet links you can upload the latest pkg file directly to https://www.virustotal.com 

Many folks use Virus Total as an automated way to gauge the safety of software before rolling out to their fleet of computers, you may wish to work with the vendors flagging your software to get it safe listed/flags removed. 

Thanks!

paul-cossey, 2024-07-23:
Thank you, appreciate you looking in to this. 

I would imagine other vendors have started to pick it up as they share detection signatures

aivarannamaa, 2024-07-24:
I got response from BitDefender:

> Our Malware Research Team has finished analyzing your case.
> The file is clean and currently not detected by our engines.

I suppose it takes time until this change propagates to VirusTotal.

paul-cossey, 2024-07-25:
Thanks, @aivarannamaa 

Yeah I guess so. Not sure how it works with other vendors changing their detections if others have stopped

aivarannamaa, 2024-08-04:
Thonny 4.1.4 is now green again in VirusTotal

closed 2024-08-04 by aivarannamaa