thonny/thonny #3260: Increase Thonny's Reconnect Tolerance

dcmcshan, 2024-07-28:
Thonny's backend gets occasionally confused when ESP32-S3 resets.  I am using the USB CDC connection, so I suspect it takes a little longer, but it clearly does come back.

Is there a way too increase Thonny's timeout tolerance for this?

Thank you!

```
>>> import machine; machine.reset()

PROBLEM IN THONNY'S BACK-END: Exception while handling 'execute_source' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

MPY: soft reboot
  1.45 boot.py
Previous reset cause: System hard reset
System soft reset
MicroPython a61c446-dirty on 2024-07-26; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
>>> import machine; machine.reset()
PROBLEM IN THONNY'S BACK-END: Exception while handling 'execute_source' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.

Process ended with exit code None.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

MPY: soft reboot
 71.46 boot.py
Previous reset cause: System hard reset
System soft reset
MicroPython a61c446-dirty on 2024-07-26; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
>>> 
```

aivarannamaa, 2024-09-03:
Sorry for the late reply.

Actually, Thonny does not even attempt to reconnect after the connection gets lost. Thonny 5.0 will have a button, which makes manual reconnection easier.

closed 2024-09-03 by aivarannamaa

dcmcshan, 2024-09-03:
Is this something I can help you beta test?

aivarannamaa, 2024-09-04:
I hope to release the first beta by the end of September