thonny/thonny #3275: PROBLEM IN THONNY'S BACK-END (class WDT of pyboard) using IDE v4.1.4

PulseDoo, 2024-08-10:
Hello
I'm having trouble running the following code using a Nucleo F401RE (MicroPython v1.23.0 on 2024-06-02) and IDE version 4.1.4.

`import pyb
import machine

green_led =  pyb.Pin('PA5', mode = pyb.Pin.OUT_PP)

#Enable watchdog timer with a timeout of 5 seconds
wdt =  machine.WDT(timeout = 5000)

sema = 0

def timer_thread(timer):
    global sema
    
    if sema == 0:
        green_led.value(not green_led.value())
        
        print("System in normal operation...")
        
        #Feed the watchdog timer to prevent it from resetting the system
        wdt.feed()

def pc13_callback(pin):
    global sema
    sema = 1
    
def main():
    global sema
    
    pyb.ExtInt('PC13', mode = pyb.ExtInt.IRQ_RISING, pull = pyb.Pin.PULL_NONE, callback = pc13_callback)
    timer1 =  pyb.Timer(1, freq = 3, mode = pyb.Timer.UP)
    #timer1 =  pyb.Timer(1, freq = 10, mode = pyb.Timer.UP) #does not work with Thonny 4.1.4, but works on v3.3.13 
    timer1.callback(timer_thread)

main()
    `
The line of code that gives the problem is the following:
``timer1 =  pyb.Timer(1, freq = 10, mode = pyb.Timer.UP)``
Using Thonny IDE v4.1.4, if I set the freq parameter to 2 or maximum 3 Hz it works, while v3.3.13 is working also for freq = 10.

The first print of the "System in normal operation..." message is aborted and the strange message "KeyboardInterrupt" appears even if I do not "interact" with the keyboard(see attached screenshot).

[image]

The problem seems independent from the Micropython version, in fact I also tried a previous version (v1.19) of the FW and found the same behavior.

Thanks in advance for your feedback

My best regards

aivarannamaa, 2024-09-03:
Serial communication with MicroPython has been rather sensitive to timing of the operations. So far such problems have been caused by bugs in the serial driver or in the communication part of MicroPython. Most likely in this case Thonny 3 wrote to the serial channel in a way that did not trigger the underlying problems. I'm afraid I can't find a timing pattern, which fits all devices on all platforms.

Thonny 5.0 will come with more advanced settings dialog for MicroPython, which allows selecting different communication methods and one of them could fix your case.

PulseDoo, 2024-09-08:
OK...thanks a lot for the feedback, then I look forward to the release of Thonny version 5.
When is it expected?

aivarannamaa, 2024-10-12:
I can't promise anything about Thonny 5.0. It has been hard to find time for Thonny lately.

closed 2024-10-12 by aivarannamaa