thonny/thonny #3281: Raspberry Pi Pico 2 and Pimoroni Plus 2 RP2350 processor USB connection not noticed

tonygo, 2024-08-15:
The current version of Thonny v4.1.4 does not recognise these boards when they are plugged in making it difficult to use with Thonny. Please add facilities for RP2350 chips so that they work like PR2040 chip on a Pico. 

The Pico 2 should work like a Pico 1

Thanks

mitag, 2024-08-17:
It was the same for me, but I don't think Thonny v14.1.4 is to blame. The newly delivered (first batch) Pimoroni Plus 2 always start in bootloader mode.

`UF2 Bootloader v1.0
Model: Raspberry Pi RP2350
Board-ID: RP2350
`

After I had installed MicroPython on a Pimoroni Plus 2, it was immediately recognised by Thonny.
[https://github.com/pimoroni/pimoroni-pico-rp2350/releases/download/v0.0.4/pico2b_rp2350-v0.0.4-pimoroni-micropython.uf2](url)

tonygo, 2024-08-17:
Does this mean that we have to load the Uf2 into the PicoPlus 2 EVERY TIME we use it or only once, as normal?
 I'm still having trouble with Thonny connecting to the Pico Plus 2.

mitag, 2024-08-17:
I may not have expressed myself clearly enough. A bootloader is not firmware. The Pimoroni Plus 2 were delivered without firmware (Micropython), so there can be no REPL prompt. When my Pimoroni Pico LiPo arrived, they were already equipped with firmware. Once you copy and paste Micropython onto the Pimoroni Plus 2, you never have to do it again. Unless you want to use a newer Micropython version, then you have to switch the microcontroller to bootloader mode (pressing Boot button while pressing Reset). Presumably the 2nd batch will also be delivered with firmware.

AeroSynth, 2024-08-30:
I am unable to save files to the Sparkfun Pro RPi 2350 Pico 2.  I get errors.  I do get the REPL prompt and can run some programs.  Just can't save them to the device.
I tried both images from micropython.org and from the Sparkfun GIT.  
Here is the error message:

```
> Traceback (most recent call last):
>   File "C:\Users\brads\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1788, in event_generate
>     handler(event)
>   File "C:\Users\brads\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1198, in update_dir_data
>     self.render_children_from_cache(msg["node_id"])
>   File "C:\Users\brads\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 457, in render_children_from_cache
>     raise RuntimeError("None data for %s" % path)
> RuntimeError: None data for /
> 
```