thonny/thonny #3283: code signing for Thonny is broken under macOS Sequoia beta

kennedye, 2024-08-15:
After downloading and installing the 4.1.4 package on a computer running macOS Sequoia 15.1, double-clicking the app icon results in an immediate crash: a `The application “Thonny” can’t be opened.` window opens and a crash report is generated:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 3616C893-B1D3-4962-8FDF-BEBBD263A2EB
CrashReporter Key:   8CBBD4AA-EF60-19CE-B9AA-8D09DB170512
Hardware Model:      Mac14,12
Process:             Python [68739]
Path:                /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:          org.python.python
Version:             3.10.11 (3.10.11)
Code Type:           ARM-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.thonny.Thonny [331752]

Date/Time:           2024-08-15 16:14:47.2350 -0700
Launch Time:         2024-08-15 16:14:47.2173 -0700
OS Version:          macOS 15.1 (24B5024e)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x1008b4b30 _dyld_start + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x0000000000000000   x2: 0x0000000000000000   x3: 0x0000000000000000
    x4: 0x0000000000000000   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000000   x9: 0x0000000000000000  x10: 0x0000000000000000  x11: 0x0000000000000000
   x12: 0x0000000000000000  x13: 0x0000000000000000  x14: 0x0000000000000000  x15: 0x0000000000000000
   x16: 0x0000000000000000  x17: 0x0000000000000000  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x0000000000000000  x21: 0x0000000000000000  x22: 0x0000000000000000  x23: 0x0000000000000000
   x24: 0x0000000000000000  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x0000000000000000   lr: 0x0000000000000000
    sp: 0x000000016f6c7ab0   pc: 0x00000001008b4b30 cpsr: 0x00001000
   far: 0x0000000000000000  esr: 0x00000000  Address size fault

Binary Images:
       0x1008b0000 -        0x100933fff dyld_path_missing (*) <68cc64d1-738b-35fd-968d-0fbd8938819f> /dyld_path_missing
       0x100738000 -        0x10073bfff main_executable_path_missing (*) <ef677878-e7fb-329d-bb8b-0f651210d52b> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------

{"app_name":"Python","timestamp":"2024-08-15 16:14:47.00 -0700","app_version":"3.10.11","slice_uuid":"ef677878-e7fb-329d-bb8b-0f651210d52b","build_version":"3.10.11","platform":0,"bundleID":"org.python.python","share_with_app_devs":1,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.1 (24B5024e)","roots_installed":0,"name":"Python","incident_id":"3616C893-B1D3-4962-8FDF-BEBBD263A2EB"}
{
  "uptime" : 270000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "Mac14,12",
  "coalitionID" : 331752,
  "osVersion" : {
[... 133 lines cut ...]
Restarting /Applications/Thonny.app
The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x6000016400f0 {Error Domain=NSPOSIXErrorDomain Code=162 "Unknown error: 162" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}
```
```
erik@m2box:~ $ /Applications/Thonny.app/Contents/MacOS/thonny 
09:17:33.361 [MainThread] INFO    thonny: Thonny version: 4.1.4
09:17:33.361 [MainThread] INFO    thonny: cwd: /Users/erik
09:17:33.361 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/MacOS/thonny', '-I', '/Applications/Thonny.app/Contents/Resources/thonnymain.py']
09:17:33.361 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
09:17:33.361 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Resources/thonnymain.py']
09:17:33.361 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
09:17:33.361 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
09:17:33.386 [MainThread] INFO    thonny.workbench: Starting Workbench
09:17:33.498 [MainThread] INFO    thonny.workbench: Creating runner
09:17:33.499 [MainThread] INFO    thonny.workbench: Start loading plugins
09:17:33.576 [MainThread] INFO    thonny.workbench: Done loading plugins
09:17:33.700 [MainThread] WARNING thonny.workbench: Could not find UI theme 'clam', switching to default
09:17:33.734 [MainThread] INFO    thonny.workbench: Opening views
09:17:33.812 [MainThread] INFO    thonny.plugins.cpython_frontend.cp_front: Creating LocalCPythonProxy
09:17:33.812 [MainThread] INFO    thonny.running: Starting the backend: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_launcher.py', '/Users/erik', "{'run.warn_module_shadowing': True}"] /Users/erik
```

I assume this is a side-effect of the revised notarization policies in Sequoia:
https://developer.apple.com/news/?id=saqachfa
but don't know whether they can even be resolved for the app. Opening through the CLI is a workaround for now but not a great user experience.

aivarannamaa, 2024-08-17:
Thank you for the report!

I was holding my breath regarding the new macOS version, as there have been unpleasant surprises before, but now it looks it's the time again.

Thonny's 4.1.4 pkg is notarized and all its content is signed (which is a prerequisite for notarization). I suspect there are changes in notarization process or maybe bugs in Sequioa. If the problem persists in the next beta, I'll re-sign and re-notarize the pkg and hopefully this fixes it.

In  the meanwhile I'll focus on getting the first beta of 5.0 out.

MartinPacker, 2024-08-28:
I have this problem, too. I'd be glad to test a fix. On Sequoia 15.0 Public Beta on a Mac Studio.

pajp, 2024-08-31:
Hi,

I don't think this is related to any changes in notarisation requirements. The Termination Reason field in the crash reports indicates that the process is being killed because it's violating its own launch constraints. Launch constraints can limit how and in what context a process can be run. I don't know enough about the Thonny build process to validate this theory, but it looks to me like Thonny is bundling a Python interpreter which has a launch constraint specifying that it may only be run from the command line, so when it's launched as a part of an app, the OS terminates it.

In my hypothesis is correct, the solution would be to remove the launch constraint from the Python interpreter before signing it, but how you do that depends on how you build & bundle the Python interpreter.

You can read more about launch constraints here: https://developer.apple.com/documentation/security/applying-launch-environment-and-library-constraints

If I have some spare time I can try to investigate this further, just thought I'd write this down in case I don't.

ibr7h, 2024-09-06:
1.	Right-click on the Thonny app in the Applications folder.
	2.	Select “Show Package Contents” to reveal the internal structure of the Thonny application bundle.
	3.	Navigate to Contents -> MacOS, where the main executable files and scripts are located.
	4.	Run Thonny directly by executing the script or binary inside this folder.

eblucas, 2024-09-06:
> ```
> 1.	Right-click on the Thonny app in the Applications folder.
> 2.	Select “Show Package Contents” to reveal the internal structure of the Thonny application bundle.
> 3.	Navigate to Contents -> MacOS, where the main executable files and scripts are located.
> 4.	Run Thonny directly by executing the script or binary inside this folder.
> or 
> ```
> 
> in Terminal /Applications/Thonny.app/Contents/MacOS/thonny

legend, it worked. thanks for sharing

aivarannamaa, 2024-09-07:
@pajp, thanks a lot for your insight! I'm currently re-signing all Python binaries before creating the installer bundle (https://github.com/thonny/thonny/blob/master/packaging/mac/sign_bundle_in_build.sh).

I'll learn more about constraints.

aivarannamaa, 2024-09-07:
Before I start digging into the topic of launch constraints, can any of you please test whether https://github.com/thonny/thonny/releases/download/v5.0.0a2/thonny-5.0.0a2.pkg works better? As Thonny 5.0 uses different location for its configuration file, it won't mess up your current configuration file.

I tried creating Sequioa VM with both VMWare and Parallels, but failed (VMWare said "no media" for my iso and Parallels failed with downloading some kind of update).

aivarannamaa, 2024-09-07:
Nevermind, I got Sequoia working in Parallels (had to install Xcode 16 beta first) and found out that the problem remains.

aivarannamaa, 2024-09-07:
_I'm recording some results, but I have another leads to investigate (I think Sequoia doesn't like that Thonny's installer does some [post-processing](https://github.com/thonny/thonny/blob/master/packaging/mac/scripts/postinstall) to the main executable, which invalidates the signature)_

Found this from the system logs:

```
AMFI: Launch Constraint Violation (enforcing), error info: c[5]p[1]m[1]e[0], (Constraint not matched) launching proc[vc: 6 pid: 952]: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python, launch type 3, failure proc [vc: 6 pid: 952]: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python
```

ChatGPT offered following explanation for the error info:

	•	error info: c[5]p[1]m[1]e[0]: These are specific error codes that map to internal constraints used by AMFI. The most relevant ones for launch constraint violations:
	•	c[5]: This could be referencing a specific class of constraint violation.
	•	p[1]: Possibly indicates an issue with the process related to its signing or permissions.
	•	m[1]: Could be related to mandatory policies (macOS might be enforcing strict conditions).
	•	e[0]: May refer to an entitlement check (this being 0 suggests no entitlement is involved).

As far as I understand, the binary in question has no launch-constraints (please correct me if I'm doing it wrong):

```
% codesign --display --launch-constraint-self - /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app
Executable=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python
% codesign --display --launch-constraint-parent - /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app
Executable=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python
% codesign --display --launch-constraint-responsible - /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app
Executable=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.12/Resources/Python.app/Contents/MacOS/Python
```

These commands were issued in Sonoma.

aivarannamaa, 2024-09-11:
Please try the new release https://github.com/thonny/thonny/releases/download/v4.1.6/thonny-4.1.6.pkg

closed 2024-09-11 by aivarannamaa

gdavids57, 2024-10-21:
Thonny 4.1.6 runs successfully on MacOS 15.0.1.