thonny/thonny #3287: INTERNAL ERROR

NkaniSM, 2024-08-22:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 317, in _cmd_run_current_script
    self.execute_current("Run")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 282, in execute_current
    script_dir = normpath_with_actual_case(os.path.dirname(filename))
  File "C:\Program Files (x86)\Thonny\lib\ntpath.py", line 221, in dirname
    return split(p)[0]
  File "C:\Program Files (x86)\Thonny\lib\ntpath.py", line 183, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

What the above error mean when trying to run code on Thonny?

BernieBranfield, 2024-09-05:
Same here, appears to be a Thonny install issue, 

Error when initializing backend

Traceback (most recent call last):
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 524, in _start_runner
    self._runner.start()
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 155, in start
    self.restart_backend(False, True)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 731, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 93, in __init__
    super().__init__(clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 32, in __init__
    super().__init__(clean, running.get_front_interpreter_for_subprocess())
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 999, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 137, in _start_background_process
    super()._start_background_process(clean=clean, extra_args=extra_args)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1093, in _start_background_process
    self._proc = subprocess.Popen(
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\subprocess.py", line 1456, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect

aivarannamaa, 2024-10-12:
Which Windows version and which Thonny version are you using?

BernieBranfield, 2024-10-12:
> Which Windows version and which Thonny version are you using?

Windows 10 Pro, version 22H2

aivarannamaa, 2024-10-12:
@BernieBranfield, which Thonny version?

After getting this error, could you please select "Open Thonny data folder" from "Tools" menu, and send me frontend.log?

BernieBranfield, 2024-10-12:
Cheers,

Bernie

BernieBranfield, 2024-10-12:
This is the frontend log after I get the error message

Cheers,

Bernie

aivarannamaa, 2024-11-26:
Sorry for the late reply! Unfortunately the attachments don't come through when replied via e-mail. You need to post it via GitHub interface.

aivarannamaa, 2025-01-15:
Please reopen if you have more information to add

closed 2025-01-15 by aivarannamaa