thonny/thonny #3290: Debug mode misses program steps

zvigrauer, 2024-08-22:
I am running the following code below. There is a object destructor code which prints something. When I run in regular mode, the message prints. When I run in debug mode, the destructor is skipped and the print is not executed.

I am not sure whether this is a big deal or not, but thought I should inform you in case this is something you want to look into.

My system is:
Thonny 4.0.1
Python 3.12.3
Ubuntu 24.04 LTS (64 bit)
CPU Intel® Core™ i7-8650U × 8

=====
class PartyAnimal:

   def __init__(self):
     self.x = 0
     print('I am constructed')

   def party(self) :
     self.x = self.x + 1
     print('So far',self.x)

   def __del__(self):
     print('I am destructed', self.x)

an = PartyAnimal()
an.party()
an.party()
an = 42
print('an contains',an)

aivarannamaa, 2024-10-12:
Python's "destructors" run after all references to the object are gone. In your script the object remains referenced by a global variable.

zvigrauer, 2024-10-12:
It's a little above my level, but are you saying that in debug mode the
variable is referenced by a global variable, but in 'regular' mode it is
not?

Could you elaborate further for my benefit? Is this behavior expected?

Thanks,

Z.

aivarannamaa, 2024-10-12:
oh, sorry, I didn't notice `an = 42` in your code.

In this case the debug mode really makes a difference, as Thonny internally keeps references to old versions of the variables (in order to support "Step back" command). Such command is not very common.