thonny/thonny #3296: Thonny not opening on M3 Macbook Air

DemonicBlaze, 2024-08-29:
Hi,

Recently upon trying to open Thonny, I have been met with these error messages. The problem has only started a few days ago and before then the application was working perfectly. I have tried reinstalling Thonny multiple times but this has not worked.

Thanks for your help.

[image]
[image]

airtonzanon, 2024-08-30:
Same issue here.

airtonzanon, 2024-08-30:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: BDF3CBC7-B214-40C6-A900-4496021D441D
CrashReporter Key:   6B9E9767-48DD-F305-A66A-3DADD2254EB6
Hardware Model:      Mac14,10
Process:             Python [5631]
Path:                /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:          org.python.python
Version:             3.10.11 (3.10.11)
Code Type:           ARM-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.thonny.Thonny [5826]

Date/Time:           2024-08-30 14:53:19.2949 +0200
Launch Time:         2024-08-30 14:53:19.2794 +0200
OS Version:          macOS 15.0 (24A5331b)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x104a0cb30 _dyld_start + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x0000000000000000   x2: 0x0000000000000000   x3: 0x0000000000000000
    x4: 0x0000000000000000   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000000   x9: 0x0000000000000000  x10: 0x0000000000000000  x11: 0x0000000000000000
   x12: 0x0000000000000000  x13: 0x0000000000000000  x14: 0x0000000000000000  x15: 0x0000000000000000
   x16: 0x0000000000000000  x17: 0x0000000000000000  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x0000000000000000  x21: 0x0000000000000000  x22: 0x0000000000000000  x23: 0x0000000000000000
   x24: 0x0000000000000000  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x0000000000000000   lr: 0x0000000000000000
    sp: 0x000000016b69fa80   pc: 0x0000000104a0cb30 cpsr: 0x00001000
   far: 0x0000000000000000  esr: 0x00000000  Address size fault

Binary Images:
       0x104a08000 -        0x104a8bfff dyld_path_missing (*) <b5d897df-d536-3668-bbad-a17119439ef0> /dyld_path_missing
       0x104760000 -        0x104763fff main_executable_path_missing (*) <ef677878-e7fb-329d-bb8b-0f651210d52b> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------

{"app_name":"Python","timestamp":"2024-08-30 14:53:19.00 +0200","app_version":"3.10.11","slice_uuid":"ef677878-e7fb-329d-bb8b-0f651210d52b","build_version":"3.10.11","platform":0,"bundleID":"org.python.python","share_with_app_devs":0,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.0 (24A5331b)","roots_installed":0,"name":"Python","incident_id":"BDF3CBC7-B214-40C6-A900-4496021D441D"}
{
  "uptime" : 33000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "Mac14,10",
  "coalitionID" : 5826,
  "osVersion" : {
    "train" : "macOS 15.0",
    "build" : "24A5331b",
[... 103 lines cut ...]

Model: Mac14,10, BootROM 11881.1.1, proc 12:8:4 processors, 32 GB, SMC 
Graphics: Apple M2 Pro, Apple M2 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Display: Q24P2W1, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Jul 26 2024 20:29:22 version 23.10.888.0.41.51.162 FWID 01-1d20bb57
IO80211_driverkit-1302.71 "IO80211_driverkit-1302.71" Aug  7 2024 03:14:49
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: RNDIS/Ethernet Gadget, Ethernet, en11
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: Logitech BRIO
USB Device: USB 2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB 2.0 BILLBOARD
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Pico
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

airtonzanon, 2024-08-30:
I see that there is an issue already open for that -> https://github.com/thonny/thonny/issues/3283

DemonicBlaze, 2024-08-30:
Thanks. I'll close this issue and keep an eye on #3283

closed 2024-08-30 by DemonicBlaze

aivarannamaa, 2024-09-11:
Please try https://github.com/thonny/thonny/releases/download/v4.1.6/thonny-4.1.6.pkg