thonny/thonny #3298: Exception while handling 'cd'

dcmcshan, 2024-09-03:
I get this error using "cd"

I do not get this with default micropython image, but I do when I overly a created FatFS image onto it.  Micropython seems to be able to use the filesystem just fine.  Wondering if you might can give me some advice on what assumptions Thonny makes wrt filesystems.

Thank you.

```
>>> %cd /lib
PROBLEM IN THONNY'S BACK-END: Exception while handling 'cd' (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

```
11:53:43.235 [MainThread] ERROR   thonny.backend: Exception while handling 'cd'
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 285, in _handle_normal_command
    response = handler(cmd)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 628, in _cmd_cd
    self._update_cwd()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 468, in _update_cwd
    super()._update_cwd()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 444, in _update_cwd
    self._cwd = self._evaluate("__thonny_helper.getcwd()")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 566, in _evaluate
    raise ManagementError("Script produced errors", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
11:53:43.238 [MainThread] ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value(__thonny_helper.getcwd())

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 42, in getcwd
OSError: [Errno 5] EIO
```

aivarannamaa, 2024-10-12:
Sorry for the late reply!

The stactrace seems to tell that function `os.getcwd()` returned an error on your device. Unfortunately I don't have a clue, what could be the reason.

dcmcshan, 2024-11-09:
Reason was entirely on my side.  I was creating a filesystem and overlaying it on vfs.  Unfortunately I was using FAT filesystem and the tools I was using were case-insensitive causing all sorts of confusion.  I didn't actually fix it.  I just went back to littlefs.

closed 2024-11-09 by dcmcshan