thonny/thonny #3302: Error when initializing backend

BernieBranfield, 2024-09-05:
Just installed Thonny on my windows PC and got the following error when opening for first time. It won't run code 

Traceback (most recent call last):
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 524, in _start_runner
    self._runner.start()
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 155, in start
    self.restart_backend(False, True)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 731, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 93, in __init__
    super().__init__(clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 32, in __init__
    super().__init__(clean, running.get_front_interpreter_for_subprocess())
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 999, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 137, in _start_background_process
    super()._start_background_process(clean=clean, extra_args=extra_args)
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1093, in _start_background_process
    self._proc = subprocess.Popen(
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\X\AppData\Local\Programs\Thonny\lib\subprocess.py", line 1456, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect

BernieBranfield, 2024-09-05:
should add, no problems when I install and run on my Mac

teachi-afk, 2024-09-19:
Hi. Same error for me. Did you solve it?

aivarannamaa, 2024-10-12:
Which Windows version and which Thonny version are you using?