thonny/thonny #3306: **Title: THONNY Unresponsive File Menu on Ubuntu 24.04 LTS**

sandip3, 2024-09-08:
**Description:**

After upgrading from Ubuntu 22.04 LTS to 24.04 LTS, Thonny's file menu becomes unresponsive. The application fails to open or save files, with the file menu not appearing or responding.

**Problems:**

- **File Menu Unresponsive:** Unable to use the file menu for opening or saving files. No dialog box appears.

**Steps to Reproduce:**

1. Install Thonny on Ubuntu 24.04 LTS.
2. Launch Thonny and attempt to use the file menu to open or save a file.

**Expected Behavior:**

The file menu should open correctly, allowing files to be opened and saved.

**Actual Behavior:**

- The file menu does not open or respond, making it impossible to open or save files.

**Environment:**

- **OS**: Ubuntu 24.04 LTS (64-bit)
- **THONNY Version**: 4.0.1
- **Python Version**: 3.12 (64-bit)
- **Zenity Version**: 4.0.1
- **Desktop Environment**: GNOME
- **Hardware**:
  - **CPU**: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
  - **RAM**: 7.5 GiB
  - **GPU**: Intel TigerLake-LP GT2 [Iris Xe Graphics], Driver: i915

**Troubleshooting Steps Taken:**

1. Restarted the laptop.
2. Tested Thonny on another PC running Ubuntu 24.04 LTS.
3. Uninstalled and reinstalled Thonny.
4. Reinstalled `zenity` and verified its version.

**Log Files/Errors:**

- Running Thonny from the terminal generates multiple errors related to `zenity`:
  ```
  01:28:29.539 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
  ...
  01:28:38.963 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
  ```

**Configuration Files:**

- **`configuration.ini`**:
  ```ini
  [general]
  configuration_creation_timestamp = 2024-09-09T01:14:02.604821
  language = en_US

  [view]
  full_screen = False
  maximize_view = False
  ui_theme = Enhanced Clam
  shellview.visible = True
  ...
  ```

- **`backend.log`**:
  ```
  01:30:13.925 INFO    thonny: Thonny version: 4.0.1
  ...
  01:30:14.002 INFO    thonny: NEW MODS ['_blake2', 'colorama', 'jedi', 'json', 'lzma', 'parso', 'pickle', 'zlib']
  ```

- **`frontend.log`**:
  ```
  01:30:12.910 INFO    thonny: Thonny version: 4.0.1
  ...
  01:31:46.018 INFO    thonny.running: Reader got EOF
  ```

**Additional Context:**

- The issue appears to be linked to `zenity`, which Thonny uses for GUI dialogs. Errors indicate `zenity` returns code `255` with the message "This option is not available," suggesting a misconfiguration or compatibility issue with Thonny or Ubuntu 24.04 LTS.
- **Zenity Configuration**: N/A
- **Temporary Workarounds**: None identified.

**Attachments:**

-  [Screencast from 2024-09-09 01-57-02.webm](https://github.com/user-attachments/assets/8f79a3b4-c4a3-449c-b02d-d5a3403760d1)

- Terminal log

``` bash
thonny 
01:57:06.009 INFO    thonny: Thonny version: 4.0.1
01:57:06.009 INFO    thonny: cwd: /home/saint
01:57:06.009 INFO    thonny: original argv: ['/usr/bin/python3', '/usr/bin/thonny']
01:57:06.010 INFO    thonny: sys.executable: /usr/bin/python3
01:57:06.010 INFO    thonny: sys.argv: ['/usr/bin/thonny']
01:57:06.010 INFO    thonny: sys.path: ['/usr/bin', '/usr/lib/python312.zip', '/usr/lib/python3.12', '/usr/lib/python3.12/lib-dynload', '/usr/local/lib/python3.12/dist-packages', '/usr/lib/python3/dist-packages']
01:57:06.010 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, safe_path=False, int_max_str_digits=4300)
01:57:06.539 INFO    thonny.plugins.cpython_frontend.cp_front: Creating LocalCPythonProxy
01:57:06.539 INFO    thonny.running: Starting the backend: ['/usr/bin/python3', '-u', '-B', '-m', 'thonny.plugins.cpython_backend.cp_launcher', '/home/saint'] /home/saint
01:57:11.851 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
01:57:12.647 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
01:57:12.647 INFO    thonny.editors: Save dialog returned None with typevariable ''
01:57:13.577 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
01:57:15.420 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
01:57:17.634 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
01:57:17.635 INFO    thonny.editors: Save dialog returned None with typevariable ''
01:57:22.854 INFO    thonny.running: Killing backend process
01:57:22.854 INFO    thonny.shell: BaseShellText._on_backend_terminated
01:57:22.857 INFO    thonny.running: Reader got EOF

```

---

selmen2004, 2024-09-08:
your thonny version is  4.0.1, that  might be the issue , concider upgrading ( latest  stable version is 4.1.4 )

sandip3, 2024-09-08:
> your thonny version is 4.0.1, that might be the issue , concider upgrading ( latest stable version is 4.1.4 )

it's Still Not working even After Update

aivarannamaa, 2024-09-11:
Zenity hasn't been a reliable component, so you could try turning off Zenity dialogs in "tools => options => general". The alternative Tk dialog is not as nice and comfortable, but more reliable. Try using Thonny's file browser (View => Files)

cubicglobe, 2024-11-11:
turning off zenity solved the problem with 4.0.1 under kubuntu 24.4. thank you!

closed 2025-01-15 by aivarannamaa