thonny/thonny #3318: MacOS Sequoia Update wont let app launch

theroof24, 2024-09-17:
The application “Thonny.app” can’t be opened.
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: FBBDDD47-61D3-4CC6-B079-56687EB0CB5C
CrashReporter Key:   A8531467-4D80-6F86-FB35-3BF8A63500FF
Hardware Model:      MacBookPro18,1
Process:             Python [6949]
Path:                /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:          org.python.python
Version:             3.10.11 (3.10.11)
Code Type:           ARM-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.thonny.Thonny [7923]

Date/Time:           2024-09-17 19:57:12.9231 -0400
Launch Time:         2024-09-17 19:57:12.8927 -0400
OS Version:          macOS 15.0 (24A335)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

Triggered by Thread:  0

Thread 0 Crashed:
0   dyld_path_missing             	       0x102570b30 _dyld_start + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x0000000000000000   x2: 0x0000000000000000   x3: 0x0000000000000000
    x4: 0x0000000000000000   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000000   x9: 0x0000000000000000  x10: 0x0000000000000000  x11: 0x0000000000000000
   x12: 0x0000000000000000  x13: 0x0000000000000000  x14: 0x0000000000000000  x15: 0x0000000000000000
   x16: 0x0000000000000000  x17: 0x0000000000000000  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x0000000000000000  x21: 0x0000000000000000  x22: 0x0000000000000000  x23: 0x0000000000000000
   x24: 0x0000000000000000  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x0000000000000000   lr: 0x0000000000000000
    sp: 0x000000016d8c3a78   pc: 0x0000000102570b30 cpsr: 0x00001000
   far: 0x0000000000000000  esr: 0x00000000  Address size fault

Binary Images:
       0x10256c000 -        0x1025effff dyld_path_missing (*) <02e3ca05-e849-31c8-a4f2-c7292420dfad> /dyld_path_missing
       0x10253c000 -        0x10253ffff main_executable_path_missing (*) <ef677878-e7fb-329d-bb8b-0f651210d52b> /main_executable_path_missing
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF
`

aivarannamaa, 2024-09-18:
Please try the latest version (4.1.6)

chrisprincenl, 2024-09-28:
that worked. 4.1.6 works on MacOS Sequoia

closed 2024-09-28 by aivarannamaa