thonny/thonny #3319: ApplePersistenceIgnoreStat problem at toplevel

wissme, 2024-09-18:
On My Mac mini running Thonny 4.1.4 and Sonoma OS, I always get on the toplevel a warning (or error ?) when evaluating the editor content.

2024-09-18 12:06:02.345 Python[12284:151839] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/c9/4v8y64sj47bbh9790lqx317c0000gr/T/org.python.python.savedState
2024-09-18 12:06:17.799 Python[12284:151839] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

aivarannamaa, 2024-09-18:
Do you get these warnings regardless of what's inside the editor?

wissme, 2024-09-18:
No, only when I use the turtle module...
    -jpr

wissme, 2024-10-06:
Hello,
Am I the only one to get this message at toplevel when using matplotlib or turtle on my Mac mini M1. Yes, maybe due to Apple Silicon, as I see that my Thonny 4.1.6 is still x86_64 based... Very annoying !

**2024-10-06 09:17:43.326 Python[55839:2942319] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/c9/4v8y64sj47bbh9790lqx317c0000gr/T/org.python.python.savedState
2024-10-06 09:17:47.293 Python[55839:2942319] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.**

Also, sometimes impossible to close the turtle or matplotlib window and obliged to STOP the program...
Thanks for us.

wissme, 2024-10-07:
I get now this kind of error with matplotlib. Abandonning Thonny for teaching, up to the correction of this bug. I was writing a book and am so sorry to go back to IDLE, I liked Thonny but support... Good luck.

TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''
>>> 2024-10-07 15:15:51.428 Python[63115:3864154] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

aivarannamaa, 2024-10-12:
Related to https://github.com/python/cpython/issues/77090