thonny/thonny #3321: Exception while handling 'editor_autocomplete' (EOFError: Ran out of input).

StoragePirate, 2024-09-19:
I can reliably reproduce breaking auto complete on two systems with `tkinter.`
```
import tkinter

MainWindow = tkinter.
```

As soon as you press the ".", you get an error.

```
Python 3.10.11 (C:\Users\MEUSER\AppData\Local\Programs\Thonny\python.exe)
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'editor_autocomplete' (EOFError: Ran out of input).
See Thonny's backend.log for more info.

Process ended with exit code 1.
```
```
...
15:39:21.795 [MainThread] INFO    thonny: TIME/MODS 0.295 Sending message highlight_occurrences_response 
15:39:22.299 [MainThread] ERROR   thonny.backend: Exception while handling 'editor_autocomplete'
Traceback (most recent call last):
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\cache.py", line 110, in wrapper
    return dct[key]
KeyError: ((), frozenset())

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\parso\cache.py", line 122, in load_module
    module_cache_item = parser_cache[hashed_grammar][file_io.path]
KeyError: WindowsPath('C:/Users/MEUSER/AppData/Local/Programs/Thonny/lib/tkinter/ttk.py')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 285, in _handle_normal_command
    response = handler(cmd)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 369, in _cmd_editor_autocomplete
    completions = jedi_utils.get_script_completions(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\jedi_utils.py", line 37, in get_script_completions
    return _export_completions(completions)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\jedi_utils.py", line 178, in _export_completions
    return [_export_completion(comp) for comp in completions]
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\jedi_utils.py", line 178, in <listcomp>
    return [_export_completion(comp) for comp in completions]
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\jedi_utils.py", line 206, in _export_completion
    name_with_symbols=_get_completion_name_with_symbols(completion),
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\thonny\jedi_utils.py", line 277, in _get_completion_name_with_symbols
    if completion.type == "param":
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\api\classes.py", line 724, in type
    return super().type
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\api\classes.py", line 190, in type
    for value in self._name.infer():
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\cache.py", line 112, in wrapper
    result = method(self, *args, **kwargs)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\names.py", line 607, in infer
    return Importer(m.inference_state, [self.string_name], m, level=self._level).follow()
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\imports.py", line 296, in follow
    return import_module_by_names(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\imports.py", line 376, in import_module_by_names
    value_set = ValueSet.from_sets([
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\imports.py", line 377, in <listcomp>
    import_module(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\plugins\__init__.py", line 21, in wrapper
    return built_functions[public_name](*args, **kwargs)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\plugins\flask.py", line 20, in wrapper
    return callback(inference_state, import_names, module_context, *args, **kwargs)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\gradual\typeshed.py", line 115, in wrapper
    python_value_set = ValueSet.from_sets(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\base_value.py", line 430, in from_sets
    for set_ in sets:
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\gradual\typeshed.py", line 116, in <genexpr>
    func(inference_state, import_names, p, sys_path,)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\imports.py", line 446, in import_module
    module = _load_python_module(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\imports.py", line 461, in _load_python_module
    module_node = inference_state.parse(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\__init__.py", line 199, in parse
    return self.parse_and_get_code(*args, **kwargs)[0]
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\jedi\inference\__init__.py", line 196, in parse_and_get_code
    return grammar.parse(code=code, path=path, file_io=file_io, **kwargs), code
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\parso\grammar.py", line 114, in parse
    module_node = load_module(self._hashed, file_io, cache_path=cache_path)
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\parso\cache.py", line 127, in load_module
    return _load_from_file_system(
  File "C:\Users\MEUSER\AppData\Local\Programs\Thonny\lib\site-packages\parso\cache.py", line 145, in _load_from_file_system
    module_cache_item = pickle.load(f)
EOFError: Ran out of input
15:39:22.304 [MainThread] INFO    thonny: TIME/MODS 0.509 Sending message ProgramOutput 
15:39:22.304 [MainThread] INFO    thonny: TIME/MODS 0.000 Sending message ProgramOutput 
```

aivarannamaa, 2024-10-12:
Thank you for the report! I'll investigate this.

Ramiruya, 2024-10-31:
Hey! I have the same problem, but it's happening at random time. This didn't happen before the kernel update.
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'editor_autocomplete' (EOFError: Ran out of input).
See Thonny's backend.log for more info.

Process ended with exit code 1.

environment: Operating System: Manjaro Linux 
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.7.2
Kernel Version: 6.10.13-3-MANJARO (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-4700HQ CPU @ 2.40GHz
Memory: 7.5 ГиБ of RAM
Graphics Processor: Mesa Intel® HD Graphics 4600
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: N750JV
System Version: 1.0

aivarannamaa, 2025-01-15:
I'll close this issue, as I'm focusing on the upcoming Thonny 5.0, which uses totally different autocompletion mechanism.

closed 2025-01-15 by aivarannamaa