thonny/thonny #3322: MicroPython array.array fails in Thonny.  OK in TerraTerm.

AeroSynth, 2024-09-20:
Have a newer ESP32-S3 N16R8.  Windows 10, Thonny 4.1.4

If I try to initialize a large array in SPIRAM, it only works with REPL in a terminal program like TerraTerm.  It does not work in Thonny either by running a program or using REPL in the Thonny shell.
Here is an example
```
import array
a = array.array('H', for _ in range(3_000_000) #create an array of 1.5M uint16.
print(a[0:10])
```

This works fine in Terra Term but hangs up Thonny and I have to reset the board.
I can get it to work in Thonny if the array size is limited to 10K, but this doesn't help.

AeroSynth, 2024-10-11:
Problem still there even if I try to allocate 200K bytes.  Works OK with 20K.  Otherwise, it completely freezes and you need to reset the board.  
On reset, this is the output:
```
MPY: soft reboot
MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Type "help()" for more information.

>>> 
```
As I mentioned previously, it works fine in a terminal program.  So Thonny is doing something wrong.
Also...if I try to import a library such as esp in the Thonny terminal, it takes a few seconds.  With Terra Term, it returns immediately.

aivarannamaa, 2024-10-12:
Thank you for the report!

Thonny is probably keeping references to some object, avoiding their garbage collection. Or it could be that the string representation of this array, which Thonny creates, is using too much memory.

I'll see what can be done.

AeroSynth, 2024-11-03:
Thanks for looking into this.  Just got a new ESP32-S3 DevKitC -1-N32R8V (32M/8M).  Problem still there.  Using latest MP 
`MicroPython v1.24.0 on 2024-10-25; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3`

AeroSynth, 2024-11-03:
Slava Ukraini