thonny/thonny #3324: Stop button does not work when using multi core (crashes sometimes)

GM-Script-Writer-62850, 2024-09-23:
Controller: PICO/RP2040
Firmware: MicroPython v1.23.0 on 2024-06-02
Screencapture: https://github.com/user-attachments/assets/875d86f2-2c7d-4056-8fe7-77ded17349da
Script:
```
from time import sleep
from _thread import start_new_thread

def core2():
	while 1:
		sleep(1)
start_new_thread(core2,())

while 1:
	sleep(1)
```
Thonny Version:
[image]

Sometimes the stop button will crash thonny under these conditions, normally if you are screen recording it it will not crash till you stop recording

aivarannamaa, 2024-10-12:
Does the problem persist in the newest Thonny version (4.1.6)?

GM-Script-Writer-62850, 2024-10-12:
i'll check, but i did notice i was unable to reproduce this on micropython 1.2

results... wow it works!!! lets try again to see if that result is reproducible... and it broke 3+ times in a row

appears to be possible for it to work, just very unlikely

console output
```
10:22:45.649 [MainThread] INFO    thonny: Thonny version: 4.1.6
10:22:45.649 [MainThread] INFO    thonny: cwd: /home/chad/Desktop/thonny/bin
10:22:45.649 [MainThread] INFO    thonny: original argv: ['/home/chad/Desktop/thonny/bin/python3.10', '-I', '-m', 'thonny']
10:22:45.649 [MainThread] INFO    thonny: sys.executable: /home/chad/Desktop/thonny/bin/python3.10
10:22:45.649 [MainThread] INFO    thonny: sys.argv: ['/home/chad/Desktop/thonny/lib/python3.10/site-packages/thonny/__main__.py']
10:22:45.649 [MainThread] INFO    thonny: sys.path: ['/home/chad/Desktop/thonny/lib/python310.zip', '/home/chad/Desktop/thonny/lib/python3.10', '/home/chad/Desktop/thonny/lib/python3.10/lib-dynload', '/home/chad/Desktop/thonny/lib/python3.10/site-packages']
10:22:45.649 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
10:22:45.861 [MainThread] INFO    thonny.workbench: Starting Workbench
10:22:45.910 [MainThread] INFO    thonny.workbench: Creating runner
10:22:45.918 [MainThread] INFO    thonny.workbench: Start loading plugins
10:22:46.159 [MainThread] INFO    thonny.workbench: Done loading plugins
10:22:46.331 [MainThread] INFO    thonny.workbench: Opening views
10:22:46.461 [MainThread] INFO    thonny.plugins.micropython.mp_front: Listing serial ports
10:22:46.587 [MainThread] INFO    thonny.shell: BaseShellText._on_backend_terminated
10:23:11.608 [MainThread] INFO    thonny.plugins.micropython.mp_front: Listing serial ports
10:23:11.657 [MainThread] INFO    thonny.running: Starting the backend: ['/home/chad/Desktop/thonny/bin/python3.10', '-s', '-u', '-B', '/home/chad/Desktop/thonny/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': False, 'validate_time': False, 'local_rtc': False}"] /home/chad/Desktop
10:23:11.724 [MainThread] INFO    thonny.shell: BaseShellText.restart(False)
10:23:26.484 [MainThread] INFO    thonny: TIME/MODS 40.854 Before Run (+201 modules)
10:23:26.485 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bisect', '_blake2', '_bz2', '_compression', '_datetime', '_hashlib', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_random', '_sha512', '_socket', '_ssl', '_struct', '_tkinter', 'array', 'base64', 'binascii', 'bisect', 'bz2', 'calendar', 'configparser', 'datetime', 'email', 'email._encoded_words', 'email._parseaddr', 'email._policybase', 'email.base64mime', 'email.charset', 'email.encoders', 'email.errors', 'email.feedparser', 'email.header', 'email.iterators', 'email.message', 'email.parser', 'email.quoprimime', 'email.utils', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'glob', 'hashlib', 'heapq', 'http', 'http.client', 'locale', 'lzma', 'math', 'ntpath', 'pathlib', 'pkgutil', 'platform', 'queue', 'quopri', 'random', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'serial.tools', 'serial.tools.list_ports', 'serial.tools.list_ports_common', 'serial.tools.list_ports_linux', 'serial.tools.list_ports_posix', 'shlex', 'shutil', 'signal', 'socket', 'ssl', 'struct', 'subprocess', 'tempfile', 'termios', 'textwrap', 'thonny.assistance', 'thonny.ast_utils', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config', 'thonny.config_ui', 'thonny.editor_helpers', 'thonny.editors', 'thonny.languages', 'thonny.memory', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.about', 'thonny.plugins.assistant_config_page', 'thonny.plugins.ast_view', 'thonny.plugins.autocomplete', 'thonny.plugins.backend', 'thonny.plugins.backend_config_page', 'thonny.plugins.base_syntax_themes', 'thonny.plugins.base_ui_themes', 'thonny.plugins.birdseye_frontend', 'thonny.plugins.calltip', 'thonny.plugins.cells', 'thonny.plugins.circuitpython', 'thonny.plugins.circuitpython.cirpy_front', 'thonny.plugins.clean_ui_themes', 'thonny.plugins.coloring', 'thonny.plugins.commenting_indenting', 'thonny.plugins.common_editing_commands', 'thonny.plugins.cpython_backend', 'thonny.plugins.cpython_backend.cp_back', 'thonny.plugins.cpython_frontend', 'thonny.plugins.cpython_frontend.cp_front', 'thonny.plugins.cpython_ssh', 'thonny.plugins.cpython_ssh.cps_front', 'thonny.plugins.debugger', 'thonny.plugins.dock_user_windows_frontend', 'thonny.plugins.editor_config_page', 'thonny.plugins.esp', 'thonny.plugins.ev3', 'thonny.plugins.event_logging', 'thonny.plugins.event_view', 'thonny.plugins.files', 'thonny.plugins.find_replace', 'thonny.plugins.general_config_page', 'thonny.plugins.goto_definition', 'thonny.plugins.heap', 'thonny.plugins.help', 'thonny.plugins.highlight_names', 'thonny.plugins.locals_marker', 'thonny.plugins.microbit', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.base_flashing_dialog', 'thonny.plugi [... cut ...]
10:24:49.936 [MainThread] INFO    thonny.running: Starting the backend: ['/home/chad/Desktop/thonny/bin/python3.10', '-s', '-u', '-B', '/home/chad/Desktop/thonny/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': False, 'validate_time': False, 'local_rtc': False}"] /home/chad/Desktop
10:24:50.008 [MainThread] INFO    thonny.shell: BaseShellText.restart(False)
```