thonny/thonny #3325: Thonny crashes since installing macOS Sequoia (Version 15.0)

nboucquey, 2024-09-23:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 4EED6459-6F24-4DC8-A97A-544BAB3385DE
CrashReporter Key:   60722915-3913-F7F1-0EF1-F64FD50D620B
Hardware Model:      Mac15,6
Process:             Python [5813]
Path:                /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:          org.python.python
Version:             3.10.11 (3.10.11)
Code Type:           ARM-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.thonny.Thonny [5228]

Date/Time:           2024-09-23 14:51:44.6105 +0200
Launch Time:         2024-09-23 14:51:44.5946 +0200
OS Version:          macOS 15.0 (24A335)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 4 Launch Constraint Violation

aivarannamaa, 2024-09-23:
Did you reinstall the same version?

randywsandberg, 2025-02-13:
Reproduced this issue today on my M1 Mac mini running macOS 15.3.1. Thonny version 4.1.7, Python 3.10.11, and Tk 8.6.13.

As suggested, I just reinstalled Thonny. Here's hoping your fix works for me too!

aivarannamaa, 2025-02-18:
I haven't seen this crash myself. I'm hoping for the fix from newer Tk version, which will be bundled in Thonny 5.0

pukulsesuatu, 2025-03-05:
same with Me even I cant install "The installer encountered and error that cause the installation to fail. Contact the software manufacturer for assistance.