thonny/thonny #3334: Crashes after 'Quit' (command=exit) in tkinter

AdgetheDancer, 2024-10-02:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 424, in execute_current
    self._wait_for_prompt(1)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 500, in _wait_for_prompt
    raise TimeoutError("Backend did not respond in time")
TimeoutError: Backend did not respond in time

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.11/tkinter/__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1035, in dispatch
    handler()
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 527, in cmd_run_current_script
    self.execute_current("Run")
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 430, in execute_current
    if self._proxy.stop_restart_kills_user_program():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'stop_restart_kills_user_program'

aivarannamaa, 2024-10-12:
Thanks for the repor!

Can you give me a sample script and exact steps for reproducing this problem?

AdgetheDancer, 2024-10-22:
Sorry for the delay, I've been away.

I've tried to replicate the problem, it's been a while and I've forgotten what I was doing.

It might have been a misunderstanding on my part.

I noticed that when a programme being written in Thonny errors (17) it doesn't reset correctly with the 'green' button inactive. I have to tap the 'red' button first to reset Thonny before preceding. It only happens if the programme was amended prior to being rerun.

I hope that helps to clarify things. Thanks for your assistance.

Alan

👴🏻🕺⌨️🚭🇬🇧

AdgetheDancer, 2024-10-24:
Here is a sample program exhibiting the error.
[Error_demo.py.pdf](https://github.com/user-attachments/files/17511062/Error_demo.py.pdf)
I hope this works for you.
[Error_demo.py.pdf](https://github.com/user-attachments/files/17511071/Error_demo.py.pdf)

aivarannamaa, 2024-11-29:
Sorry for the late response!

Unfortunately I can't reproduce this problem. Which Thonny version are you using?