thonny/thonny #3336: ESP32 flashing failure always get error 2?

Jibun-no-Kage, 2024-10-06:
ESP32 flashing failure always get error 2?

```
C:\Users\jibun\AppData\Local\Programs\Thonny\python.exe -u -m esptool --port COM8 erase_flash
esptool.py v4.4
Serial port COM8
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Erase command returned with error code 2C:\Users\jibun\AppData\Local\Programs\Thonny\python.exe -u -m esptool --port COM8 erase_flash
esptool.py v4.4
Serial port COM8
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Erase command returned with error code 2C:\Users\jibun\AppData\Local\Programs\Thonny\python.exe -u -m esptool --port COM8 erase_flash
esptool.py v4.4
Serial port COM8
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Erase command returned with error code 2C:\Users\jibun\AppData\Local\Programs\Thonny\python.exe -u -m esptool --port COM8 erase_flash
esptool.py v4.4
Serial port COM8
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Erase command returned with error code 2C:\Users\jibun\AppData\Local\Programs\Thonny\python.exe -u -m esptool --port COM8 erase_flash
esptool.py v4.4
Serial port COM8
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Erase command returned with error code 2
```

ESP32 module currently has tasmota installed, no issue flash tasmota to this ESP32 in the past.

Jibun-no-Kage, 2024-10-06:
I was able to install MicroPython 1.22 but not 1.3?

aivarannamaa, 2024-10-12:
Did you try different esptool parameters (flash mode, Flash size)?

Jibun-no-Kage, 2024-10-12:
I did try various options.  What is odd is that I have flash many modules via various methods, even use Thonny to flash in the past with success.  What worked... as version 1.22 consistently, but I still cannot get 1.3 to load at all.  Is there anything in version 1.3 that could be some type of hardware limitation issue?  I suspect not, but the fact remains, 1.22 ESP32 generic flashes and 1.3 ESP32 generic does not.

aivarannamaa, 2024-11-26:
I suggest trying with plain esptool and if still having problems, ask in a MicroPython forum.

closed 2024-11-26 by aivarannamaa

Jibun-no-Kage, 2024-11-26:
Ultimately, I may do that as well... I have not had time to get back to this, but should have time in the next couple of days.  I have a Linux image, explicitly to use esptool, as my last resort.  Once in a while I have IoT devices that just refuse to flash Tasmota with various other tools at times.

What is interesting is that it is the Mini D1 (blue boards) that seem to get me the most issues, usually the various ESP32 modules are painless, but not in this case.