thonny/thonny #3340: how to config 2 Thonny instances shortcuts, one is for Pico, the other one is for ESP32?

bk111, 2024-10-08:
my son used to Pico, I used to ESP32 on the same PC. Can I config 2 shortcuts for them?

Ivoz, 2024-10-12:
As an alternative, I think it would be very nice to be able to switch the python interpreter from a dropdown in the main interface, rather than having to go through the menu. I often switch pythons between programming MCUs and on the computer, or switch to different virtualenvs, and it is a bit cumbersome going through the settings tabs each time.

aivarannamaa, 2024-10-12:
Thonny 5.0 will have a `--profile` command line option, which allows running it with different set of settings. I'll consider adding menu items for different profiles as well.

qeijm, 2024-11-27:
I didn't want to start a new 'issue' but this is similar to what I want. When I have a RPi running using say COM3, when I launch a second instance of Thonny **with the intent to locally try out a code fragment** , Thonny opens a 2nd instance using COM3 which sometimes halts the first instance. What I would like to see is a command line to open a local python instance of Thonny using local Python interpreter. I don't necessarily need a way to call every available interpreter but at least local vs not local (which could be numerous). Thanks @aivarannamaa