thonny/thonny #3342: Update to Python 3.13

TrangOul, 2024-10-11:
[Python 3.13 has just been released](https://www.python.org/downloads/). One of the additions are [improved error messages](https://docs.python.org/3/whatsnew/3.13.html#improved-error-messages), which would surely be helpful for beginners.

The authon mentioned that 5.0 will come with Python 3.12 (#2931). Please strongly consider upgrading to 3.13.

aivarannamaa, 2024-10-12:
As the launch of Thonny 5.0 has been delayed, I'll bundle it with Python 3.13.

OldSubSailor, 2025-05-31:
Perhaps Python 3.12 would be good enough for the current version for now?

speedy-10, 2026-01-31:
> As the launch of Thonny 5.0 has been delayed, I'll bundle it with Python 3.13.

When will Thonny 5.0 be released?