thonny/thonny #3343: Install micropython on esp32 -  esp32tool module location issue

tobes, 2024-10-11:
When installing micropython on an esp32 using thonny it asked me to install esptool via Tool > Manage plug-ins which I did.

When I then tried to again install micropython I had a problem and thonny gave me an error.

The installation cannot find the esptool module as it is expecting it to be available but it was installed  in `~/.config/Thonny/plugins/lib/python3.12/site-packages/` which python couldn't find.

Running `pip install esptool` in the cli installed the package to `~/.local/lib/python3.12/site-packages` and allowed thonny to then work as expected.

Although this was an easy issue for me to resolve it would not be so for a novice user.

Thonny 4.1.4, Fedora 40, Python 3.12.6

aivarannamaa, 2024-10-12:
Thanks for this information! I'll try to reproduce and fix it later.

LoggerJunge, 2024-11-15:
Thonny 4.0.2, Fedora 39, Python 3.12.7  Same problem and many thanks for the solution.  Namaste, Logger-Boy

tobes, 2025-02-15:
Update for Fedora users with this issue

I have now upgraded to Fedora 41 and the problem is worse.  Python version has been upgraded to `3.13.2`

Thonny no longer can access `esptool` from `~/.local/lib/python3.13/site-packages` .  The reason is that `/usr/bin/thonny` contains the following `#! /usr/bin/python3 -sP` this prevents site packages being in the path.  Changing this line to `#! /usr/bin/python3 -P` allows site packages and allows thonny to run with the manual pip install workaround.

The actual issue remains the same that Thonny does not allow esptool to be installed, it cannot find it in pypi either which I'm not sure was a problem previously.