thonny/thonny #3353: Thonny 5.0.0.0a2 macOS, package installer error, pip

Oppfields, 2024-10-18:
Apple Studio M1 Max
macOS 14.6.1
Thonny 5.0.0.0a2

Thonny installed as single user (~/Applications), opened from package contents ( right click, open with Terminal).

When trying to install a local package (zip) file the following warning message appears and the process fails to complete:

`Error Command '['/Users/xxxx/Library/Caches/pipkin/workspaces/0a08dc960c413736e943412e64c37352/bin/python3', '-I', '-m', 'pip', '--disable-pip-version-check', 'install', '--no-warn-script-location', 'pip==24.0']' returned non-zero exit status 1.`

abc1111111-sudo, 2024-11-01:
SSL Isn't working for me either.

aivarannamaa, 2024-11-02:
Please ignore this release. It remained visible by accident.

closed 2024-11-02 by aivarannamaa