thonny/thonny #3360: Broken shell on Pi Pico 2 RP2350

frindaddy, 2024-10-22:
On startup, Thonny shell displays `INTERNAL ERROR, got '' instead of 'OK'` and does not start the Pico shell.

This happens regardless of what Micropython version I run on the Pico 2, and regardless of installing with Thonny vs drag and drop .uf.

Micropython is installed correctly, as I can communicate with the Pi Pico 2 and toggle the onboard LED through a PuTTy serial terminal at 115200 baud.

Pictures attached:
[image]
[image]
Terminal output:
`OK`
`☻1 BackendEvent(event_type='HideTrailingOutput', text='MicroPython v1.24.0-preview.460.g6328958a3 on 2024-10-22; Raspberry Pi Pico2 with RP2350-RISCV\r\nType "help()" for more information.\r\n', sequence='BackendEvent', cwd=None, sys_path=None, lib_dirs=None)`
`☻1 ToplevelResponse(event_type='ToplevelResponse', cwd='/', welcome_text='MicroPython v1.24.0-preview.460.g6328958a3 on 2024-10-22; Raspberry Pi Pico2 with RP2350-RISCV\r\nType "help()" for more information.\r\n', sequence='ToplevelResponse', sys_path=['', '.frozen', '/lib'], lib_dirs=['/lib'])`

aivarannamaa, 2025-01-15:
Sorry for the late reply! If you are still interested in getting this sorted out, then please send me the backend.log from Tools => Options => Open Thonny data folder.

frindaddy, 2025-01-17:
I fixed this problem just a few days ago by installing Thonny via pip rather than the installer found on the website. Somehow, that managed to get everything working.

aivarannamaa, 2025-01-17:
Great!

closed 2025-01-17 by aivarannamaa