thonny/thonny #3368: cannot flash Waveshare ESP32-S3-ZERO properly

stanelie, 2024-11-06:
Hello.
If I flash micropython on my ESP32-S3-ZERO, I get an error in REPL when the device starts : 
`ESP_ERR_FLASH_NOT_INITIALISED`
The only way I got working was by flashing it manually : 
```
esptool.py --chip esp32-S3 --port /dev/cu.usbmodem14201 erase_flash
esptool.py --chip esp32-S3 --port /dev/cu.usbmodem14201 write_flash -z 0 ESP32_GENERIC_S3-FLASH_4M-20241025-v1.24.0.bin  
```

I got the firmware on the micropython website (select the .bin file) : 
https://micropython.org/download/ESP32_GENERIC_S3/

Maybe this bug explains https://github.com/thonny/thonny/issues/3044

aivarannamaa, 2025-01-15:
Do you mean you get different results when flashing via Thonny's esptool GUI vs flashing from command line?

stanelie, 2025-01-15:
Yes.
It does not work through Thonny, it works if I flash with the Circuitpython online tool (https://circuitpython.org/downloads)