thonny/thonny #3377: Directory sensitive file transfer/sync

dcmcshan, 2024-11-15:
Is there some way to have file transfer be directory sensitive?  Like I have a file tree that has /lib/io and I'd like files to transfer to /lib/io on the target?  Would save me a lot of clicks, as well as some missteps (where I accidentally load files into wrong directory)

Alternatively, it would be super helpful to just have a straight up sync between "This computer" and "MicroPython device"

ppamment, 2025-01-01:
+1 on this.
I very frequently hit upload on a file only to realise I had not cd'd to the correct directory and it ends up in the root. It seems very natural for $projectroot/lib/foo/bar.py to upload to /lib/foo/bar.py by default rather than $cwd/bar.py.

A full diff/sync would be a very nice thing to have, but that's obviously a significantly more substantial feature request

aivarannamaa, 2025-01-15:
Thanks for the suggestion! I'll consider adding this feature.