thonny/thonny #3383: menu bar cant be clicked

anak-ut, 2024-11-23:
im using pop_os! with wayland

i have installed thonny from ubuntu repository and flatpak
but i cant do select any menu bar option on application windows.

aivarannamaa, 2025-01-15:
Sorry for the late reply!

If you are still having this problem -- does it happen right after launching Thonny or after some activities?

kevinrineer, 2025-09-29:
Hi folks, I think I've experienced the same issue with respect to the menu on linux. pop_os! is based on Ubuntu, so this may be related.

```
cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.3 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```
Running in an xsession:

```
env |& grep -i xorg

XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu-xorg:/etc/xdg
DESKTOP_SESSION=ubuntu-xorg
XDG_SESSION_DESKTOP=ubuntu-xorg
XDG_DATA_DIRS=/usr/share/ubuntu-xorg:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
GDMSESSION=ubuntu-xorg
```

Help > About
```
Ubuntu 24.04.3 (64-bit)
Python 3.12.3. (64-bit)
Tk 8.6.14
```

Trying to open or save files results in what seem to be Zenity errors.
```
$ cat thonny.log

12:55:57.241 INFO    thonny: Thonny version: 4.0.1
12:55:57.242 INFO    thonny: cwd: /home/local/rineerk-local
12:55:57.242 INFO    thonny: original argv: ['/usr/bin/python3', '/usr/bin/thonny']
12:55:57.242 INFO    thonny: sys.executable: /usr/bin/python3
12:55:57.242 INFO    thonny: sys.argv: ['/usr/bin/thonny']
12:55:57.242 INFO    thonny: sys.path: ['/usr/bin', '/usr/lib/python312.zip', '/usr/lib/python3.12', '/usr/lib/python3.12/lib-dynload', '/usr/local/lib/python3.12/dist-packages', '/usr/lib/python3/dist-packages']
12:55:57.242 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, safe_path=False, int_max_str_digits=4300)
12:55:57.688 INFO    thonny.plugins.cpython_frontend.cp_front: Creating LocalCPythonProxy
12:55:57.688 INFO    thonny.running: Starting the backend: ['/usr/bin/python3', '-u', '-B', '-m', 'thonny.plugins.cpython_backend.cp_launcher', '/home/local/rineerk-local'] /home/local/rineerk-local
12:56:00.417 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
12:56:01.614 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
12:56:01.614 INFO    thonny.editors: Save dialog returned None with typevariable ''
12:56:10.363 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
12:56:13.627 INFO    thonny.running: Killing backend process
12:56:13.627 INFO    thonny.shell: BaseShellText._on_backend_terminated
12:56:13.630 INFO    thonny.running: Reader got EOF
```

I'm unfamiliar with your project and I don't exactly have time right now, so I'll end my comment here.