thonny/thonny #3386: Thonny makes the screen go black and then crashes

DiamondMattia, 2024-11-28:
Hi, when I use Thonny on my linux machine (an old Dell Vostro 1510), the computer hangs, the screen goes black and then Thonny crashes or stays open for a while and crashes after some moments. This doesn't happen regularly: sometimes it happens when I start the software, sometimes when I open a file, sometimes when I save, sometimes when I scroll through a file and sometimes randomly.
I tried using both the apt version of Thonny and the snap version with the same result.
I get the same result with both X11 and wayland.

SPECS:
OS: KDE neon 6.2 x86_64 (but i've also got the same problem with Linux Mint 22)
Kernel: 6.8.0-49-generic
CPU: Intel Core 2 Duo T5870
GPU: Intel Mobile GM965/GL960
Memory: 3906MiB

aivarannamaa, 2024-11-29:
Which Thonny version are you using?

DiamondMattia, 2024-11-29:
the latest (4.1.6)

closed 2025-01-15 by aivarannamaa