thonny/thonny #3390: Debgging tools greyed out..why?

denukvt, 2024-12-05:
I am running the 'happy birthday' code, which runs fine.
However I cannot run any of the debuggers because theyare all greyed out.

I'm very new to this, so uinable to find any solution.

I am running Win 11 and Thonney 4.1.6 if that helps.

denukvt, 2024-12-05:
[image]
[image]

Is this any good? Never done a screenshot before. 

Heres a file that is a picture from "Paint"

[image]
ot before.

aivarannamaa, 2024-12-05:
I now see you are using MicroPython. Thonny does not have a debugger for MicroPython.

denukvt, 2024-12-05:
OK my mistake (as I suspected !)

Does it have the debuggers for C?

aivarannamaa, 2024-12-05:
No. Only for CPython.