thonny/thonny #3392: Tools > Manage Packages fails to load any packages

smorrisonsmith, 2024-12-06:
When navigating to Tools > Manage Packages in Thonny 4.1.6, the expected list of packages does not load. Instead, the interface remains blank, preventing users from searching for or installing packages via the GUI. Installation must be done manually through the terminal using pip. Problem is reproducible with other Macs with the following chips: M1, M2, M3, M4, and also on a PC running Windows 11.

lilahdelussey, 2024-12-06:
i keep getting told command not found: pip. what do I do? Im on mac. im freaking out because I have a project due tonight and i cant get anything downloaded.

smorrisonsmith, 2024-12-06:
> i keep getting told command not found: pip. what do I do? Im on mac. im freaking out because I have a project due tonight and i cant get anything downloaded.

Did you try pip3? Thorny comes with python 3.10 so I think pip3 command may work.

jshaffstall, 2024-12-06:
I'm also seeing this issue on various Windows machines and on various networks.

chrisib, 2024-12-07:
Searching for packages like e.g. `ssd1306` gives me results from micropython-lib, but I don't see any PyPI results.

I'm a maintainer of [this project](https://github.com/Allen-Synthesis/EuroPi) and we rely on Thonny for installing packages from PyPI on the Raspberry Pi Pico.

Searching directly on the site shows the package exists (0.15.0): https://pypi.org/search/?q=micropython-europi.

But searching for `europi`, `micropython-europi`, or `micropython_europi` in Thonny's package manager isn't giving any results anymore.

I suspect it's an issue between Thonny's search window and PyPI's search API? That's speculative on my part, but it seems to match the results I'm (not) seeing.

tiyaswoohoo, 2024-12-07:
when i search for pygame, the search comes empty. i have tried installing pygame from shell too but no luck.

aivarannamaa, 2024-12-08:
It looks like Thonny's approach for searching PyPI has been disabled: https://discuss.python.org/t/fastly-interfering-pypi-search/73597/

ryentzer, 2024-12-08:
For others who need to install packages in the meantime, I was able to do so with the following steps:

1. Select Tools > Open system shell

You should see that the shell is using the correct Thonny python version. You can confirm with `which python3`

2. Search for the package you want on https://pypi.org
3. Install it by running `python3 -m pip install your_package`. Replace "your_package" with your package. :)

bryanosborne, 2024-12-08:
> For others who need to install packages in the meantime, I was able to do so with the following steps:
>
> 1. Select Tools > Open system shell
>
> You should see that the shell is using the correct Thonny [...]

Sadly, this did not work for ARM64 (MacOS). When importing pandas an incompatible architecture message is given. :-(

A complete uninstall/reinstall of thonny and python followed by the above pip install yields:
Traceback (most recent call last):
  File "/Users/xxxxx/Documents/test.py", line 1, in <module>
    import pandas as pd
  File "/Users/xxxxx/Library/Python/3.10/lib/python/site-packages/pandas/__init__.py", line 19, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

Thonny is broken.

harrma, 2024-12-10:
same, it just shows a blue loading circle infinitely

sairamjadala, 2024-12-12:
When navigating to Tools > Manage Packages in Thonny 4.1.6, the expected list of packages does not load. Instead, the interface remains blank, preventing users from searching for or installing packages via the GUI. Installation must be done manually through the terminal using pip. Problem is reproducible with running Windows 11 in laptop

.

aivarannamaa, 2024-12-16:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.7

closed 2024-12-16 by aivarannamaa

CM630, 2026-02-16:
The issue is **not** resolved in v4.1.7, behaves as described by @smorrisonsmith.
The mouse cursor changes to a spinning (progress) wheel and stays forever.
Windows 11 64 bit on amd64.

I also tried what @bryanosborne proposed - the packages are isntalled, but _Thonny_ does not see them.