thonny/thonny #3397: Importing matplotlib to Thonny

TMartina75, 2024-12-11:
Hi!
My students can not import and ad matplotlib into their program by adding it to the packages. I was able to do in on my computer a week or two ago. They need it to plot diagrames. How can we solve this problem.

Best regards, Martina Nilsen

aivarannamaa, 2024-12-15:
Please see #3401

closed 2024-12-15 by aivarannamaa

aivarannamaa, 2024-12-16:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.7

snakethumpergit, 2025-09-03:
I'm getting this error on Thonny 4.1.4 when i try to import matplotlib:

Traceback (most recent call last):
  File "/home/david/Documents/TestMatPlotLib.py", line 8, in <module>
    import matplotlib
  File "/usr/lib/python3/dist-packages/matplotlib/__init__.py", line 113, in <module>
    from . import _api, _version, cbook, _docstring, rcsetup
  File "/usr/lib/python3/dist-packages/matplotlib/rcsetup.py", line 27, in <module>
    from matplotlib.colors import Colormap, is_color_like
  File "/usr/lib/python3/dist-packages/matplotlib/colors.py", line 51, in <module>
    from PIL import Image
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: /lib/aarch64-linux-gnu/libwebp.so.7: invalid ELF header

I checked and I have libwebp.so.6 but not 7

I'm a relative newbie to Pi. I've checked and matplotlib is installed. There seems to be a later version of Thonny which might fix this, but i have no idea how to install it without a venv. (i dont need or want a venv)

Can anyone point me in the right direction?